LAStools/LAStools #6: L-GPL license question

mousebird, 2015-10-07:
I'd like to use LAStools to interface to my [geospatial display library](http://mousebird.github.io/WhirlyGlobe/).  My library is open source, so it presents me no real problem, but I know my users won't pay attention to the license of anything I use.  I have to pay attention for them.

My question is if you have an L-GPL exemption for static linking on iOS.  The dynamic replacement requirement of that license is really impossible to meet on shipping iOS apps.

mousebird, 2015-10-07:
Just found [this](https://groups.google.com/forum/#!topic/lastools/Bs9YBXzXXcg).  Answers my question.

closed 2015-10-07 by mousebird

rapidlasso, 2015-10-07:
The LASzip library (not the LASlib library or any of the LAStools) have a
static linking exception for exactly that purpose.

http://github.com/LAStools/LAStools/blob/master/LASzip/LICENSE.txt

Hope this helps ...

Martin