LAStools/LAStools #8: Is lasreader sensitive to 64 vs 32 bit compilation of the library?

plitkey, 2016-02-03:
I have used the LASlib library in my read and write routines. Now I had a data that I could not open. LAStools and PulseWaves opened it nicely. My reader gave library error that file signature is not LASF, it is. I can read the file with my older reader that is not using the LASlib library. The error comes in both windows and linux versions of my Matlab reader, both use 64 bit compilations of LASlib.

rapidlasso, 2016-02-03:
No idea. I have not yet compiled LASlib on 64 bits ...

ksshannon, 2016-02-03:
I've compiled and used on amd64 with gcc/linux.  Can you post the file?

esilvia, 2016-02-03:
It might depend on your compiler and how you're reading the file signature.
I've read that some compilers read char as a 16-bit character by default if
you have unicode enabled, which results in you reading more bytes than you
intended. You can check whether this is happening to you by checking "pos()"
after reading the signature. You should be at 4 if it read correctly.

Evon

esilvia, 2016-02-03:
I should note that I have not observed this issue when compiled using amd64
on MSVC2010.

Evon

plitkey, 2016-02-04:
I have found the error - In my code as usual :-) Thanks for ideas!

closed 2016-02-04 by plitkey