LAStools/LAStools #12: Compilation fails: undefined reference to `LASreadOpener::open(char const*)'

Ylannl, 2016-05-31:
I want to compile LAStools. Tried it on two different platforms that worked before (OS X 10.11 and Ubuntu 14.04), but now both fail with a similar error(OS X uses `clang` and Ubuntu `gcc`):

OS X:
```
g++  -O3 -Wall lasexample.o -llas -o lasexample  -L../lib -I/usr/include -I../../LASzip/src -I../inc
Undefined symbols for architecture x86_64:
  "LASreadOpener::open(char const*)", referenced from:
      LASindex::append(char const*) const in liblas.a(lasindex.o)
     (maybe you meant: __ZN13LASreadOpener4openEPKcb)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Linux:
```
g++  -O3 -Wall -Wno-deprecated -DNDEBUG  laszip.o geoprojectionconverter.o -llas -o laszip  -L../LASlib/lib  -I../LASzip/src -I../LASlib/inc 
../LASlib/lib/liblas.a(lasindex.o): In function `LASindex::append(char const*) const':
lasindex.cpp:(.text+0xaf8): undefined reference to `LASreadOpener::open(char const*)'
../LASlib/lib/liblas.a(lasreadermerged.o): In function `LASreaderMerged::open()':
lasreadermerged.cpp:(.text+0x2ec5): undefined reference to `LASreaderLASrescale::LASreaderLASrescale(double, double, double)'
../LASlib/lib/liblas.a(lasreaderbuffered.o): In function `LASreaderBuffered::open()':
lasreaderbuffered.cpp:(.text+0x111e): undefined reference to `LASreadOpener::open(char const*)'
lasreaderbuffered.cpp:(.text+0x2272): undefined reference to `LASreadOpener::open(char const*)'
collect2: error: ld returned 1 exit status
```

esilvia, 2016-06-02:
Perhaps your project is missing lasindex.cpp, lasreadermerged.cpp, and
lasreaderbuffered.cpp, probably from your Makefile? Your issue sounds
similar to this issue:
http://stackoverflow.com/questions/6978241/undefined-reference-to-in-g-cpp

Taking a stab in the dark here, but it sounds similar enough to explore.

Evon

KermMartian, 2016-06-03:
I can confirm this, on Ubuntu 15.10. Another machine with Ubuntu 14.10 builds the same project correctly. It appears to be defined in lasreader.o (`0000000000001ce0 T _ZN13LASreadOpener4openEPKcb`), but not in the final liblas.a (`U _ZN13LASreadOpener4openEPKcb`). Some optimization running amok, perhaps?

KermMartian, 2016-06-09:
*bump* I discovered that I had an old fork (arsf/LAStools) installed, as well as this trunk (LAStools/LAStools) linked from a local directory. The header from arsf/LAStools was being used with the library binary from LAStools/LAStools, to the predictably incongruous results. This may be the case for you as well? I used c++filt to compare the type signatures carefully, and noticed that the library was offering a set of possible arguments to open() that didn't match the include file.

Ylannl, 2016-06-13:
Thanks for you thoughts guys. I just re-cloned the project on both machines and now it compiles without errors.

closed 2016-06-13 by Ylannl