LAStools/LAStools #27: lasinfo crash using -stdin on .laz file

kjk, 2017-03-23:
The file I'm using: https://coast.noaa.gov/htdata/lidar1_z/geoid12a/data/1434/20121114_ct40.laz

The code is latest from git, compiled on mac by running `make`.

When I run without `-stdin` option, everything works:
```
$ ./bin/lasinfo /Volumes/hd/data/lidar/source/noaa/noaa_1434/geoid12a/data/1434/20121114_ct40.laz
lasinfo (170313) report for /Volumes/hd/data/lidar/source/noaa/noaa_1434/geoid12a/data/1434/20121114_ct40.laz
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             161
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.2
  system identifier:          'ALS60'
  generating software:        'Fugro Horizons LAS Lib v1.14'
  file creation day/year:     348/2012
  header size:                227
  offset to point data:       713
  number var. length records: 5
  point data format:          1
  point data record length:   28
  number of point records:    2608
  number of points by return: 2302 218 79 9 0
  scale factor x y z:         0.000000099999966 0.0000001 0.001
  offset x y z:               0 0 0
  min x y z:                  -73.380750787446232 41.0891644 -1.268
  max x y z:                  -73.380201487634963 41.0898975 25.251
variable length header record 1 of 5:
  reserved             43707
  user ID              'NIIRS10'
  record ID            4
  length after header  10
  description          'NIIRS10 Timestamp'
variable length header record 2 of 5:
  reserved             43707
  user ID              'NIIRS10'
  record ID            1
  length after header  26
  description          'NIIRS10 Tile Index'
variable length header record 3 of 5:
  reserved             43707
  user ID              'FugroHorizons'
  record ID            0
  length after header  128
  description          'FHZNInfo'
variable length header record 4 of 5:
  reserved             43707
  user ID              'NOAA_CSC'
  record ID            129
  length after header  2
  description          'Sort Order'
variable length header record 5 of 5:
  reserved             43707
  user ID              'LASF_Projection'
  record ID            34735
  length after header  48
  description          'Projection Parameters'
    GeoKeyDirectoryTag version 1.1.0 number of keys 5
      key 1024 tiff_tag_location 0 count 1 value_offset 2 - GTModelTypeGeoKey: ModelTypeGeographic
      key 2048 tiff_tag_location 0 count 1 value_offset 4269 - GeographicTypeGeoKey: GCS_NAD83
      key 2054 tiff_tag_location 0 count 1 value_offset 9102 - GeogAngularUnitsGeoKey: Angular_Degree
      key 4096 tiff_tag_location 0 count 1 value_offset 5103 - VerticalCSTypeGeoKey: VertCS_North_American_Vertical_Datum_1988
      key 4099 tiff_tag_location 0 count 1 value_offset 9001 - VerticalUnitsGeoKey: Linear_Meter
the header is followed by 2 user-defined bytes
LASzip compression (version 2.1r0 c2 50000): POINT10 2 GPSTIME11 2
reporting minimum and maximum for all LAS point record entries ...
  X          -733807760 -733802267
  Y           410891644  410898975
  Z               -1268      25251
  intensity           0         79
  return_number       1          4
  number_of_returns   1          4
  edge_of_flight_line 0          0
  scan_direction_flag 0          1
  classification      1          8
  scan_angle_rank     1          3
  user_data         204        210
  point_source_ID    38         38
  gps_time 37011731.478286 37011732.577899
WARNING: range violates GPS week time specified by global encoding bit 0
number of first returns:        2302
number of intermediate returns: 87
number of last returns:         2304
number of single returns:       2085
overview over number of returns of given pulse: 2085 278 213 32 0 0 0
histogram of classification of points:
             563  unclassified (1)
            1720  ground (2)
             325  keypoint (8)
```

When I use `-stdin` option on the same file, it crashes:
```
$ ./bin/lasinfo -stdin </Volumes/hd/data/lidar/source/noaa/noaa_1434/geoid12a/data/1434/20121114_ct40.laz
lasinfo (170313) report for piped input
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             161
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.2
  system identifier:          'ALS60'
  generating software:        'Fugro Horizons LAS Lib v1.14'
  file creation day/year:     348/2012
  header size:                227
  offset to point data:       713
  number var. length records: 5
  point data format:          1
  point data record length:   28
  number of point records:    2608
  number of points by return: 2302 218 79 9 0
  scale factor x y z:         0.000000099999966 0.0000001 0.001
  offset x y z:               0 0 0
  min x y z:                  -73.380750787446232 41.0891644 -1.268
  max x y z:                  -73.380201487634963 41.0898975 25.251
variable length header record 1 of 5:
  reserved             43707
  user ID              'NIIRS10'
  record ID            4
  length after header  10
  description          'NIIRS10 Timestamp'
variable length header record 2 of 5:
  reserved             43707
  user ID              'NIIRS10'
  record ID            1
  length after header  26
  description          'NIIRS10 Tile Index'
variable length header record 3 of 5:
  reserved             43707
  user ID              'FugroHorizons'
  record ID            0
  length after header  128
  description          'FHZNInfo'
variable length header record 4 of 5:
  reserved             43707
  user ID              'NOAA_CSC'
  record ID            129
  length after header  2
  description          'Sort Order'
variable length header record 5 of 5:
  reserved             43707
  user ID              'LASF_Projection'
  record ID            34735
  length after header  48
  description          'Projection Parameters'
    GeoKeyDirectoryTag version 1.1.0 number of keys 5
      key 1024 tiff_tag_location 0 count 1 value_offset 2 - GTModelTypeGeoKey: ModelTypeGeographic
      key 2048 tiff_tag_location 0 count 1 value_offset 4269 - GeographicTypeGeoKey: GCS_NAD83
      key 2054 tiff_tag_location 0 count 1 value_offset 9102 - GeogAngularUnitsGeoKey: Angular_Degree
      key 4096 tiff_tag_location 0 count 1 value_offset 5103 - VerticalCSTypeGeoKey: VertCS_North_American_Vertical_Datum_1988
      key 4099 tiff_tag_location 0 count 1 value_offset 9001 - VerticalUnitsGeoKey: Linear_Meter
the header is followed by 2 user-defined bytes
LASzip compression (version 2.1r0 c2 50000): POINT10 2 GPSTIME11 2
reporting minimum and maximum for all LAS point record entries ...
Segmentation fault: 11
```

rapidlasso, 2017-03-23:
Nice find. I just pushed a tiny fix to LASzip/src/lasreadpoint.cpp that seems to resolve this stdin read issue.

kjk, 2017-03-23:
Thanks, it looks like it's fixed.

closed 2017-03-24 by rapidlasso