LAStools/LAStools #33: Compile error in laszip_dll.cpp

perhallstroem, 2017-07-14:
> /build/LAStools/LASzip/src/laszip_dll.cpp: In function 'laszip_I32 laszip_read_point(laszip_POINTER)':
> /build/LAStools/LASzip/src/laszip_dll.cpp:4220:91: error: lvalue required as left operand of assignment
>         ((U16)(point->rgb[3])) = *((U16*)(point->extra_bytes + laszip_dll->start_NIR_band));

See comment on file in commit 7e7a3f3e6dc5c3b9a6bc7bbc81e4092d8455987a

rapidlasso, 2017-07-14:
Do you know how to fix it? My compilers are all happy.

dabulla, 2017-07-19:
I also had this error (gcc-Version 7.1.1 20170630).

rapidlasso, 2017-07-19:
Do you know how to fix it? You see what I intend to do. A two-byte unsigned ints is supposed to be assigned to the fourth position of point->rgb[] that is an array of  two-byte unsigned ints.

perhallstroem, 2017-08-17:
Sorry for the delay. See my inline comment in that same commit.

rapidlasso, 2017-08-17:
Hopefully fixed by [this commit](http://github.com/LAStools/LAStools/commit/0f054902ff97538025493bb8e68e291871d9eb5b).

rapidlasso, 2017-09-15:
Seems all are happy now.

closed 2017-09-15 by rapidlasso