LAStools/LAStools #34: Compiling using Visual Studio 2015

jakka30, 2017-08-28:
Hello all,

Did anyone try to upgrade VC++6.0 projects to Visual Studio 2015? I tried to do it and am getting errors and it failed to upgrade projects.  Please let me if any one has done this before successfully.

Thanks,
Suresh

esilvia, 2017-09-07:
I haven't been able to upgrade from the packaged v6.0 projects for a couple
years now. Personally, I've been repurposing an old SLN file that I
converted back in 2014(ish) using MSVS2010 by manually deleting and
replacing the source files each time. It's not perfect and I think the
example executables are broken as a result, but it works.

Evon

n-mam, 2017-09-19:
for the original vc6 project files (*.dsp) could you export the corresponding windows nmake mak file and add those to the repo ? Once the dsw (workspace) is loaded with all the dsp's (projects); you could highlight each project one-by-one on the left hand side pane; then click on "Project" menu option and select "Export Makefile". This should generate mak file for the corresponding project which nmake can consume to do a command line build. That way we could try build using command line using more recent VS.

esilvia, 2017-09-20:
This is an archived version of my solution file from VS2010 that you might be able to repurpose for your application. As I said, it's pretty outdated, but the bones are good and if you replace the guts with the new source code it should work with minimal modification.

[LASzip_to_lastools.zip](https://github.com/LAStools/LAStools/files/1318784/LASzip_to_lastools.zip)

n-mam, 2017-09-21:
Thanks Evon. I'll try this out today. 

Meanwhile, I created a VS2015 solution, added the latest laslib and laszip sources. With this, I was only able to build the static libs for laslib and laszip. 

laszip.dll builds fine with LASZIPDLL_EXPORTS defined but the link for laslib.dll fails as lasreader.cpp picks up single argument seek_next() calls as the conditional LASZIPDLL_EXPORTS branches are not in place there. Besides this, earlier I also had to change few win32 APIs to explicitly use the ANSI version instead. Like: the code has GetModuleHandleEx and this by default gets resolve to the "W" suffixed  GetModuleHandleExW when the string argument being passed are ANSI strings. Another, things which I had to change was replacing the #include of hash_map  with unordered_map.. I'll share all the changes for your review soon.

Neelabh

rapidlasso, 2017-09-21:
Don't mix laszip_dll.cpp and the code from LASlib. They are either or. The LASzip DLL uses laszip_dll.cpp to create the interface. The LASlib library uses lasreader.cpp and laswriter.cpp ...

rapidlasso, 2018-12-29:
There are CMake files now that do the trick for other VC++ Compilers (I have tested MSVC2010, MSVC2012, and MSVC2017).

closed 2018-12-29 by rapidlasso