LAStools/LAStools #36: lax_index seek_next number of arguments?

plitkey, 2017-08-30:
I'm having problems compiling the library:

1>..\laszip_dll.cpp(4391): error C2660: 'LASindex::seek_next' : function does not take 2 arguments
I downloaded yesterday and checked  the  line in this master:
4931   while (laszip_dll->lax_index->seek_next(laszip_dll->reader, laszip_dll->p_count))

how to fix this?

rapidlasso, 2017-08-30:
Are you mixing and matching different versions? This part of the code has
not been touched in a long long time ... or is this the first time you are
trying to use the DLL API instead of the LASlib API?

plitkey, 2017-08-30:
Yes, I am trying to mix things, I took all source files from laslib and laszip and all headers and made a project of those on visual studio 2010 (the projects did not open succesfully in VS update thingy). I wanted to make a static library that I could put inside the matlab mex function.

rapidlasso, 2017-08-30:
There is no need to mix laszip_dll.cpp with LASlib. The laszip_dll.cpp is
basically implementing the same use of LASzip that you find in
LASreader_las.cpp and LASwriter_las.cpp of LASlib. You only need one of the
two: either LASlib or laszip_dll.cpp.

plitkey, 2017-08-30:
Thanks! I had a successful build without laszip_dll.cpp :-)

closed 2017-08-30 by plitkey