LAStools/LAStools #38: undefined reference to `LASreadPoint::LASreadPoint()`

adi64, 2017-10-19:
I tried building the project on my Linux machine (gcc 7.2.0), but it fails when building `lasunzipper`.
My procedure was as instructed, aka `cd LAStools` and `make`.

```
[...]
make[2]: Entering directory '[...]/LAStools/LASzip/example'
g++  -O3 -Wall -Wno-strict-aliasing laszippertest.o ../src/laszip.o ../src/laszipper.o ../src/lasunzipper.o ../src/lasreadpoint.o ../src/lasreaditemcompressed_v1.o ../src/lasreaditemcompressed_v2.o ../src/lasreaditemcompressed_v3.o ../src/laswritepoint.o  ../src/laswriteitemcompressed_v1.o ../src/laswriteitemcompressed_v2.o ../src/laswriteitemcompressed_v3.o ../src/integercompressor.o ../src/arithmeticdecoder.o ../src/arithmeticencoder.o ../src/arithmeticmodel.o -o laszippertest   -I/usr/include -I../src
../src/lasunzipper.o: In function `LASunzipper::open(_IO_FILE*, LASzip const*)':
lasunzipper.cpp:(.text+0x7c): undefined reference to `LASreadPoint::LASreadPoint()'
../src/lasunzipper.o: In function `LASunzipper::open(std::istream&, LASzip const*)':
lasunzipper.cpp:(.text+0x293): undefined reference to `LASreadPoint::LASreadPoint()'
collect2: error: ld returned 1 exit status
```
However, these methods are declared in `LASzip/src/lasreadpoint.hpp` and defined in `LASzip/src/lasreadpoint.cpp`. Maybe the Makefile is broken?

dhdl, 2017-11-06:
For me ./laszippertest builds without any problems (also ArchLinux).

adi64, 2017-11-08:
Just checked, it also works for me again. Thanks!

closed 2017-11-08 by adi64