LAStools/LAStools #42: Las2las intensity errors converting .pts to .laz

pdean, 2017-11-22:
Using the precompiled las2las.exe, I'm getting nonsense intensity errors.  When I recompile with mingw, the errors go away.
(I've reported this on the google forum but had no response)

.pts file looks like this with intensities in range -2048..2047
--------
181204304
11661.296646 70527.929642 9.934738 -640
11661.184311 70527.957626 9.873734 -366
11661.173080 70527.958023 9.862625 -314
....
-----------
Batch file is this
las2las -i ../pts/westmid.pts ^
        -v ^
        -iparse xyzi ^
        -ipts ^
        -rescale 0.0001 0.0001 0.0001 ^
        -olaz ^
        -odir . 

Error messages see  https://s18.postimg.org/l22muizy1/las2las_errors.png
eg
WARNING: intensity 5.93139e+007 is out of range of unsigned short
WARNING: intensity 9.55139e+007 is out of range of unsigned short
WARNING: intensity 1.19514e+008 is out of range of unsigned short
WARNING: intensity -7.60098e+007 is out of range of unsigned short
WARNING: intensity 3.31139e+007 is out of range of unsigned short
.........
In file of 180 miillion points, 29 errors. 
In file of 600 million, many more, I gave up and recompiled.  
I don't have microsoft compilers, I used msys2/mingw64 and added -static -static-libgcc -static-libstdc++ to the linker options in the Makefile.
I'm on Windows 10.

pdean, 2017-11-23:
For somebody to confirm my results, you need a very large .pts file to test.  I wrote a simple program that will generate a 9gig .pts file.  Just pipe the output into a file.
#include <stdlib.h>
#include <stdio.h>

int main() {
    int i,j,k,I;
    double x0,y0,x,y,z;
    srand(355113);
    x0=10000.0;
    y0=20000.0;
    k = 15000;

    printf("%d\n",k*k);
    for (i=0; i<k; i++) {
        x=x0+i*0.01+rand()*.01/RAND_MAX;
        for (j=0; j<k; j++) {
            y=y0+j*0.01+rand()*.01/RAND_MAX;
            z=100.0*rand()/RAND_MAX;
            I=(rand()&4095)-2048;
            printf("%f %f %f %d\n",x,y,z,I);
        }
    }
}

rapidlasso, 2018-12-29:
Has this been resolved? I don't recall if we had follow up on this ...