LAStools/LAStools #43: Incorrect 'context'

Wanwannodao, 2017-11-28:
The variable `context` is declared here, 
```cpp:laswritepoint.cpp
BOOL LASwritePoint::write(const U8 * const * point)
{
  U32 i;
  U32 context = 0;
```
and is set to `current_context` only if 'scanner channel' has changed from that of the previous encoded point.
```cpp
if (changed_values & (1 << 6))
{
    U32 diff = dec_channel_returns_XY->decodeSymbol(contexts[current_context].m_scanner_channel); // curr = last + (sym + 1)
    U32 scanner_channel = (current_context + diff + 1) % 4;
    // maybe create and init entropy models and integer compressors
    if (contexts[scanner_channel].unused)
    {
      // create and init entropy models and integer decompressors
      createAndInitModelsAndDecompressors(scanner_channel, contexts[current_context].last_item);
    }
    // switch context to current scanner channel
    current_context = scanner_channel;
    context = current_context; // the POINT14 reader sets context for all other items

    // get last for new context
    last_item = contexts[current_context].last_item;
    ((LASpoint14*)last_item)->scanner_channel = scanner_channel;
}
```
However, this is neither a static variable nor a global variable, and thus is initialized to zero whenever a point is encoded, which means that, for example,  `context` is still 0 even if `current_context` of POINT14 and the previous 'scanner channel' are 1.
Because all subsequent layers use `context` set by POINT14, their `current_context`s will be set to this incorrect context (scanner channel).
This is not the same behavior as the past `global_current_context`, right ?

rapidlasso, 2017-11-30:
I'm in Fiji and sick as a dog so I cannot debug this right now - even though it would be a critical issue,

rapidlasso, 2017-11-30:
No. I think you are wrong. The context is set (on every restart of a point block) in the init() function that is always called before and read() or write() calls.

```
  /* set scanner channel as current context */

  current_context = ((LASpoint14*)item)->scanner_channel;
  context = current_context; // the POINT14 reader sets context for all other items

  /* create and init models and decompressors */

  createAndInitModelsAndDecompressors(current_context, item);
```

rapidlasso, 2017-11-30:
Oh shit. I think you may be right. Of course I cannot simply merge this pull request as it would destroy all past native-compressed LAS 1.4 content.

rapidlasso, 2017-11-30:
May I ask how you found this?

Wanwannodao, 2017-11-30:
yes, I can understand the logic...
Let me explain in more detail. Think about the case `scanner_channel` is always 1 for simplicity.
As you said, when the first point of a chunk is processed, `context` is correctly set in `init()` of POINT14 so all  other `writers_compressed`'s `current_context` are initialized to 1 . so far so good.
However, the problem occurs during encoding the second.
First, `context` is initialized to 0 (at the top of `LASwirtePoint::write`) and passed to POINT14.
```cpp
  if (writers)
  {
    for (i = 0; i < num_writers; i++)
    {
      writers[i]->write(point[i], context); // context = 0 because context is a local variable
    }
  }
```
After POINT14 layer,  `context` is still 0, not 1, cuz `(changed_values & (1 << 6))` is false.
(we assume `scanner_channel` is always 1)
```cpp
  if (changed_values & (1 << 6))
  {
    ... 
    // switch context to current scanner channel
    current_context = scanner_channel;
    context = current_context; // the POINT14 writer sets context for all other items
  }
```
A subsequent layer, say, BYTE14 will get, as its argument, `context`(=0) that is not updated by POINT14.
The bug occurs here,
```cpp
  if (current_context != context)
  {
    current_context = context; // all other items use context set by POINT14 writer
    ...
  }
```
Thanks to the correct initialization, `current_context` is 1 at the beginning of `write()` of BYTE14; however,   `current_context != context` is **true** (`context` is 0).
As a result, `current_context` will be incorrectly updated with 0 although the correct value is 1.
am i wrong...?

Wanwannodao, 2017-11-30:
ah, i see...

For a reason, I dumped  `current_context`, `context` and so on, and i observed a strange transition of `current_context` like 0->1->0 while `scanner_channel` changed as 0->1->1.
Finally, i though this as a bug by comparing the behavior of the current `context` with the past `global_current_context` in source code.

rapidlasso, 2017-12-01:
It does mean that the compressor does currently not take advantage of context switching and performance is lower on LAS 1.4 input with populated scanner channels. I messed up when I "carefully" removed the `global_current_context ` to create a per instance `context ` that would allow multiple writers to operate simultaneously (like lastile or lassplit). Which secret government agency are you working for?

esilvia, 2017-12-05:
I was about to integrate native 1.4 compression in my app. Does this mean I should wait on implementation, or is this benign enough that nothing's actually broken on the data side?

rapidlasso, 2017-12-05:
The current version will be supported in future versions. There may be a release with a more optimized compressor addressing the above (new *.compressed_v4.cpp/hpp readers) . But tons of native LAS 1.4 content based on the current scheme is being produced at NOAA and USGS. So it's not going away ... and the issue only affects heavily attributed data from multi-beam systems anyways. The core POINT14 compression is not affected.

esilvia, 2017-12-05:
Thanks Martin! That's really great to know. Hope you're starting to feel better, and sorry you have to deal with this headache.

Wanwannodao, 2017-12-17:
There is a similar bug even if the context is correct.

`last_item` must be switched as well as `current_context` whenever a context switch occurs.
A wrong `last_item` is currently used and overwritten when context switching.

```cpp
  U8* last_item = contexts[current_context].last_item;

  // check for context switch

  if (current_context != context)
  {
    current_context = context; // all other items use context set by POINT14 writer
    if (contexts[current_context].unused)
    {
      createAndInitModelsAndCompressors(current_context, last_item);
      last_item = contexts[current_context].last_item;
    }
  }
```

rapidlasso, 2017-12-17:
I've noticed that too while debugging your initial finds. There will be a _v4 release of the "native" LASzip item compressors once I'm done travelling and have some time to scrutinize this. Note that this only affects LAS files containing multi-beam LiDAR with populated scanner channel *and* the point type is higher than 6 such that additional items beyond POINT14 are used.

rapidlasso, 2017-12-17:
May I ask who you are? Do we know each other? How come you are scrutinizing the code to this depth?

Wanwannodao, 2017-12-21:
No, I'm just a passing engineer who has recently been doing something like porting LASzip, so i had to dissect it.

rapidlasso, 2017-12-29:
So. I've started preparing to improve the context as it was meant by releasing a version 4 of all compressed item readers that is part of the DLL and the lib. In a year (or so) we can then switch the compressed writer to produce this new version 4. The idea is to propagate the v4 readers long before producing v4 content to assure a smooth transition. For most content I have played with there was little to no difference in compression anyways.

Maybe you can check my changes?

https://github.com/LAStools/LAStools/commit/5bf547b8b43703f06de501d20f8faa5c82c81d3e
and
https://github.com/LAStools/LAStools/commit/815cc386e692fafac277f7631909a27bb85d77b4

to actually create v4 content you will need to change this line of code to set all versions to a 4 instead of a 3.

https://github.com/LAStools/LAStools/blob/815cc386e692fafac277f7631909a27bb85d77b4/LASzip/src/laszip.cpp#L649

Wanwannodao, 2018-01-09:
I briefly checked the v4 readers/writers, and they seem to work fine.
Thanks for your hard work !

manfred-brands, 2018-12-27:
Whilst writing nunit tests for my C# version, I found that the LASreadItemCompressed_RGBIR14_v3::read has a bug that it cannot always read what it compresses. This is dependent on those cases where `last_item` is not always set to the right one.
It determines changes relative to the `last_item` set, but then instead of updating that, it explicitly updates `contexts[current_context].last_item[3]`. This is correct for the v4 decoders, but due to the bug in the v3 decoders, this is not always the same as `last_item`.

It can be fixed by changing the code to look like the below (I left in original code commented out):

[image]

rapidlasso, 2018-12-27:
The v4 was created to fix semantically incorrect v3. But you say there is an actual bug? Do you have a sample LAS file with NIR channel that compresses with laszip.exe and then incorrectly decompresses with laszip.exe?

manfred-brands, 2018-12-27:
Yes, it is easy to create (in code). The only fields that need to be set are a changing scanner channel and the near infra red.
[Nir14.txt](https://github.com/LAStools/LAStools/files/2712117/Nir14.txt)

`txt2las -set_point_type 8 -parse lI -i NIR14.txt -o NIR14.las`

The input file is: scanner channel, NIR:

0 0
1 1
2 2
3 3
0 4
1 5
2 6
3 7
0 8
1 9

When compressing with las2las and decompressing with las2las, then las2txt gives:

0 0
1 1
2 2
3 3
0 4
1 9
2 14
3 19
0 23
1 27

rapidlasso, 2018-12-27:
Thanks for this bug discovery and bug fix report. It seems that there is little content with multiple scanner channels that has the NIR channel populated. This commit should fix it, can you confirm?

https://github.com/LAStools/LAStools/commit/2382d225f409b2eef97c2a1a61c6e79322b01a9b

manfred-brands, 2018-12-27:
Yes, this fixes the problem and indeed it is hard to find LAS files that have filled in anything but XYZ (RGB).

manfred-brands, 2018-12-27:
There is one more small 'bug' in that LASlib refuses to decode files with WAVEPACKET14 version 4 compression.
You update the error message to include version 4, but not the test.

[image]

manfred-brands, 2018-12-28:
Finding any 3 party LAS files with WavePackets is impossible. Let alone finding WDP files.
I found this issue because my unit test files could not be read by LAStools.

The attached file is a Format9.laz converted using 'base64' to allow me to attach it here.
Don't expect any sensible WavePackets.

[Format9.base64.txt](https://github.com/LAStools/LAStools/files/2714400/Format9.base64.txt)

manfred-brands, 2018-12-28:
We haven't had contact previously, Although my employer has a company/world wide license from you.
A lot of our new developments are in .NET and there was no decent (native) .NET version of LASlib. I therefore made a C# version capable of reading/writing LAS/LAZ. I had some quiet time over the Christmas period to update the library to include the new LAZ 1.4 compression.
Out of principle, all our libraries have full unit tests. In this case I have to be sure it is 100% compatible with LASlib. These unit tests are against the C# version and would not be of any use in LASLib itself.