LAStools/LAStools #54: Memory error when writing extra bytes

Jean-Romain, 2018-05-13:
When writing extra bytes `laslib` have a memory error. This might be reproduced with your [example file](https://github.com/LAStools/LAStools/blob/master/LASlib/example/lasexample_write_only_with_extra_bytes.cpp) running the example with `valgrind`

```
valgrind --tool=memcheck --leak-check=full ./lasexample_write_only_with_extra_bytes test.las
```

This is the report I get. I guess the error comes either from `laswriter_las.cpp:74` or `laswriter_las.cpp:920`

```
==15274== Memcheck, a memory error detector
==15274== Copyright (C) 2002-2015, and GNU GPL'd, by Julian Seward et al.
==15274== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==15274== Command: ./lasexample_write_only_with_extra_bytes test.las
==15274== 
==15274== Syscall param write(buf) points to uninitialised byte(s)
==15274==    at 0x54D52C0: __write_nocancel (syscall-template.S:84)
==15274==    by 0x5456BFE: _IO_file_write@@GLIBC_2.2.5 (fileops.c:1263)
==15274==    by 0x5458408: new_do_write (fileops.c:518)
==15274==    by 0x5458408: _IO_do_write@@GLIBC_2.2.5 (fileops.c:494)
==15274==    by 0x54591CE: _IO_switch_to_get_mode (genops.c:177)
==15274==    by 0x5456509: _IO_file_seekoff@@GLIBC_2.2.5 (fileops.c:1012)
==15274==    by 0x5454E68: fseeko (fseeko.c:36)
==15274==    by 0x41447E: ByteStreamOutFile::seek(long long) (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274==    by 0x40A154: LASwriterLAS::update_header(LASheader const*, bool, bool) (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274==    by 0x401D3C: main (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274==  Address 0x5ac3918 is 280 bytes inside a block of size 4,096 alloc'd
==15274==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==15274==    by 0x544B1D4: _IO_file_doallocate (filedoalloc.c:127)
==15274==    by 0x544DFFA: setvbuf (iosetvbuf.c:60)
==15274==    by 0x411C3B: LASwriterLAS::open(char const*, LASheader const*, unsigned int, int, int, int) (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274==    by 0x407473: LASwriteOpener::open(LASheader const*) (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274==    by 0x401AB8: main (in /home/jr/Téléchargements/LAStools/LASlib/example/lasexample_write_only_with_extra_bytes)
==15274== 
==15274== 
==15274== HEAP SUMMARY:
==15274==     in use at exit: 72,704 bytes in 1 blocks
==15274==   total heap usage: 25 allocs, 24 frees, 79,041 bytes allocated
==15274== 
==15274== LEAK SUMMARY:
==15274==    definitely lost: 0 bytes in 0 blocks
==15274==    indirectly lost: 0 bytes in 0 blocks
==15274==      possibly lost: 0 bytes in 0 blocks
==15274==    still reachable: 72,704 bytes in 1 blocks
==15274==         suppressed: 0 bytes in 0 blocks
==15274== Reachable blocks (those to which a pointer was found) are not shown.
==15274== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==15274== 
==15274== For counts of detected and suppressed errors, rerun with: -v
==15274== Use --track-origins=yes to see where uninitialised values come from
==15274== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

rapidlasso, 2018-05-13:
Interesting. Seems to be something happening wrong with the memory allocated by setvbuf(). If you add 

setvbuf(file, NULL, _IOFBF, 0);

call just before the

  if (file)
  {
    setvbuf(file, NULL, _IOFBF, 0);
    fclose(file);
    file = 0;
  }

does that memory leak go away?

Jean-Romain, 2018-05-13:
No it does not. 

Btw it is not a memory leak but a memory error. It is about writing uninitialized bytes or something like that. This kind of error may appear when writing uninitialized bytes due to padding alignment for example. In your case I found that everywhere I checked the memory was initialized.

```cpp
memset(this, 0, sizeof(LASattribute));
```

So, it might be something else. As a minimal reproducible example this code raise the error but if you remove  `lasheader.update_extra_bytes_vlr();` it does not.

```cpp
#include <time.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "laswriter.hpp"

int main(int argc, char *argv[])
{
  LASwriteOpener laswriteopener;
  laswriteopener.set_file_name("test.las");

  LASheader lasheader;
  LASattribute attribute(2, "echo width", "full width at half maximum [ns]", 1);
  lasheader.add_attribute(attribute);
  lasheader.update_extra_bytes_vlr();
  lasheader.point_data_record_length += lasheader.get_attributes_size();

  LASwriter* laswriter = laswriteopener.open(&lasheader);
  laswriter->close();
  delete laswriter;

  return 0;
}

closed 2018-10-24 by Jean-Romain