LAStools/LAStools #61: Example on how to use lax file?

neverwiredhouse, 2018-09-02:
I want to grab small subsections of lidar and analyze them,

I have an `.las` and a `.lasx` but I assume the `lasx` is just a `lax`.

Was looking through other issues, examples, and the code itself, still a little iffy on how to approach reading data. I've been reading entire files and just snagging the ones within my bounding box, but it's drastically slower then I think I can achieve and I need to do this roughly 20k times, so I need to improve the algorithm or it will take weeks as opposed to what I would assume could be just hours.

rapidlasso, 2018-09-02:
Those *.lasx files are proprietary closed clone crap by ESRI and not the open source *.lax files produced by lasindex.

rapidlasso, 2018-09-02:
We give free consultation only via our public user forum that you can find at

http://groups.google.com/group/lastools

Please include as much information about your data, density, data volume, covered area, lasinfo reports, expected area-of-interest query pattern, ... as possible.

closed 2018-12-29 by rapidlasso