LAStools/LAStools #67: Writing LAZ with compatibility_mode enabled doesn't update point record size

esilvia, 2019-01-16:
Hello! Normally I write all LAZ 1.4 files using the version-3 compressor, but I currently have a client that doesn't yet support the native 1.4 compression.

When I attempt to write my LAZ 1.4 file (PDRF=6) with compatibility mode enabled, the point record size never gets updated. `laszip_open_writer()` used to do this, but it doesn't seem to do so any more.

The point record size of my file should be 33 (28+5), but it's still 30 when written.

It seems that the number of "compatibility" extrabytes gets computed correctly around here...

https://github.com/LAStools/LAStools/blob/master/LASzip/src/laszip.cpp#L505

...and it seems to be used for the actual point record compression, but at no point does the header object get updated. Previous versions of LASzip would modify the header to set the version to LAS 1.2 and the point record size to 33.

Is this user error or a bug?

I'm running version 180617 if that matters.

rapidlasso, 2019-01-17:
Can you provide a small example that I can run in the debugger? Are you using laszip.exe?

esilvia, 2019-01-21:
I'm pretty much just following EXAMPLE_SEVEN but I'll see what I can put together. First, though, I should update to the latest and make sure you didn't already fix this... I just noticed a couple related tweaks in a couple commits.

rapidlasso, 2019-01-27:
I just tried EXAMPLE_SEVEN and it worked.

I noticed that in laszip.exe the '-compatible' or '-no_native' switch did not work when the input file was compressed in the native LAS 1.4 format. It only worked when going via uncompressed LAS. That's fixed in the version 190127 that I just released. Maybe the same issue for your DLL use?