LAStools/LAStools #68: Extra Bytes fields offset wrong

manfred-brands, 2019-01-25:
We have a las file created according to the Topo-Bathy Lidar. This dictates a triple field for sigma_xyz.
I saw discussion on removing this, but haven't seen any updated specification.

You removed support for these fields in on 14-Sep2018.
However, even if I now only want to extract a field after that particular field, its offset is off.

Can we at least in: LASattribute.get_size() return the correct size multiplied with the dimensions?
That way all other fields can be extracted correctly.

[matlab.las.base64.txt](https://github.com/LAStools/LAStools/files/2794877/matlab.las.base64.txt)

manfred-brands, 2019-01-25:
Te following gives the correct output for field 234

    if (data_type)
    {
      const U32 size_table[10] = { 1, 1, 2, 2, 4, 4, 8, 8, 4, 8 };
      U32 type = get_type();
-      return size_table[type];
+      int dim = 1 + ((I32)data_type - 1)/10;
+     return size_table[type] * dim;
    }

rapidlasso, 2019-01-26:
Okay. This commit should fix that: https://github.com/LAStools/LAStools/commit/3a4a25b770dc53dcd1ef7c1013a5876737462dd2

Please use "nicer" offsets. Gets you fewer WARNINGs in lasinfo and looks more professional:

instead of
-71.690406400000001 21.863001799999999 -41.719999999999999
use
-71.7 21.9 -41.7
or similar

closed 2019-01-26 by rapidlasso

manfred-brands, 2019-01-27:
Thanks,

The offset and min/max values in the header are rounded to the scale factor (in this case 0.0000001).
Lasinfo prints all values using 15 decimal places, but a double only has 15 digits of resolution.
With 2 digits before the decimal point, that leaves in this case only 13 decimal digits which would have printed as -71.6904064 21.8630018 -41.72

Are you suggesting to round these values to about 1 decimals place regardless of specified resolution?

rapidlasso, 2019-01-27:
The offset are meant to get you into the vicinity of the data. I set mine to a rounded value that is close to the nearest number per 10 million scale factors.

https://github.com/LAStools/LAStools/blob/master/LASlib/src/lasreader_las.cpp#L1606

manfred-brands, 2019-01-29:
Thanks. 

Note that you have another place where you use offset/scale even if not set. (las2txt::print_attribute)
You have an 'OR' in the test but always use both.

The current spec says to set these fields to zero if not used.

```
    if (header->attributes[index].has_scale() || header->attributes[index].has_offset())
    {
      F64 temp_d = header->attributes[index].scale[0]*value + header->attributes[index].offset[0];
      lidardouble2string(printstring, temp_d, header->attributes[index].scale[0]);
      fprintf(file, "%s", printstring);
    }
```

You could simplify your code a lot and remove all repeated run time tests if you would set the scale to 1 and offset to 0 for those attributes that don't define them in the init_attributes method.
Something like:
````
      if (!attributes[i].has_scale())
      {
          attributes[i].scale[0] = attributes[i].scale[1] = attributes[i].scale[1] = 1.0;
      }
````

rapidlasso, 2019-01-29:
True. That is another bug. This commit fixes it: https://github.com/LAStools/LAStools/commit/18471441333cc84aa9f7a8c0ae6537286714f909

I chose not to modify the scale[3] and offset[3] values as they would get written modified back to file and that might interfere with user intention (temporary disabling of particular scales or offsets).