LAStools/LAStools #69: e572las.exe error when trying to split scans

ollydoyle, 2019-03-14:
Using the command:
e572las -v -i "D:\Working\Fla\2.e57" -o Split.las -split_scans

I get repeats of this:
processing scan 1 of 33 ...
  contains grid of 4895 by 2448 equaling 11982960 points
  has quaternion (0.68435,-0.00868469,-0.00205505,0.729099) which is applied
  has translation (2001.24,1003.56,102.434) which is applied
no x coordinates for scan 0. skipping ...

I have viewed the file in other softwares fine. Any ideas? maybe it a not combatible version of e57?

rapidlasso, 2019-03-14:
Hello. Could you share a small (!!!) sample e57 file that exhibits the same "no x coordinates for scan 0. skipping ..." message? Ideally by posting a link here on in the LAStools user forum at:

https://groups.google.com/group/lastools

ollydoyle, 2019-03-19:
i try making a post but i dont know if it work on that forum

here is a small file that causes the same error
https://wetransfer.com/downloads/3909df882ebea25b10a31c4fe6fe1dd820190319092631/148308

rapidlasso, 2020-02-23:
Did we every fix that? I don't recall ... ?