LAStools/LAStools #75: undefined reference to `LASignore::LASignore()'

mwil, 2019-08-24:
Both the make and cmake builds fail in the current master with the following error:
```
[...]
cd example && make
make[2]: Entering directory '/home/me/src/lastools-git/LASlib/example'
g++  -c -O3 -Wall -Wno-strict-aliasing -Wno-unused-result -I/usr/include -I../../LASzip/src -I../inc lasexample.cpp -o lasexample.o
g++  -O3 -Wall -Wno-strict-aliasing -Wno-unused-result lasexample.o -llas -o lasexample  -L../lib -I/usr/include -I../../LASzip/src -I../inc
../lib/liblas.a(lasreader.o): In function `LASreadOpener::parse(int, char**, bool)':
lasreader.cpp:(.text+0x58af): undefined reference to `LASignore::parse(int&, int, char**)'
lasreader.cpp:(.text+0x6091): undefined reference to `LASignore::LASignore()'
lasreader.cpp:(.text+0x63b8): undefined reference to `LASignore::LASignore()'
lasreader.cpp:(.text+0x63c0): undefined reference to `LASignore::usage() const'
lasreader.cpp:(.text+0x63cd): undefined reference to `LASignore::~LASignore()'
lasreader.cpp:(.text+0x6e63): undefined reference to `LASignore::~LASignore()'
lasreader.cpp:(.text+0x785b): undefined reference to `LASignore::~LASignore()'
collect2: error: ld returned 1 exit status
Makefile:19: recipe for target 'lasexample' failed
make[2]: *** [lasexample] Error 1
make[2]: Leaving directory '/home/me/src/lastools-git/LASlib/example'
Makefile:2: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/me/src/lastools-git/LASlib'
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2
```

I was able to fix it by including lasignore.o in LASlib/src/Makefile to the OBJ_LAS.

In addition, the signature of a function in header lasignore.hpp is wrong. It reads
`BOOL unparse(CHAR* string) const;`
but the return type is actually `I32`.

rapidlasso, 2019-08-24:
Thanks from the Danube Delta in Romania. I didn't get around to update the Makefile. My bad. Thanks for the other bug report also. Should be fixed.

closed 2019-08-24 by rapidlasso

manfred-brands, 2019-08-26:
@rapidlasso I had to add it manually to laslib.vcxproj after running CMake.

Also for VS 2017 could you add the following to .gitignore?

CMakefiles
CMakeCache.txt
*.cmake
*.vcxproj*
*.sln