LAStools/LAStools #89: LASpoint::is_zero() and LASpoint::zero() don't agree

pauljnixon, 2020-02-19:
LASpoint::is_zero() checks the first few fields via pointer manipulation to make sure they are all zeroed out.  

But LASpoint::zero() doesn't zero out return_number or number_of_returns (both set to 1).  

Since is_zero() is not used within LAStools I'm guessing the issue is there and that zero() is correct.  

I plan to write a pull request to fix is_zero(), but I'm not sure if return_number and number_of_returns should be checked to see if they are 1, or just unchecked.  

I'm using MSVC2017, in case the byte packing is the issue here.

rapidlasso, 2020-02-20:
Actually I think it's just poor naming. The zero() command should maybe be called reset() and the is_zero() function should be eliminated (as it's not used anywhere). What are you using it for?

pauljnixon, 2020-02-20:
I've received an LAS with over a million points at 0, 0, 0.  It's a problem for the KD-tree I'm reading the points into, to have so many points in any one spot, so I want to drop them before that stage.  I'll just write my own function to test the points.

closed 2020-02-20 by pauljnixon

rapidlasso, 2020-02-23:
I am deleting the unused (and poorly named) function 'is_zero()' from the library ...