LAStools/LAStools #93: Error reading PLY files

spono, 2020-04-30:
Hi,
I'm accessing a PLY file generated with CloudCompare but I get this message as a blocking error. I guess it comes from `LASlib`, even if I do not imagine to what it might be related.
The point cloud is a 3x3 metres scan derived from photogrammetry with local coordinates, and 0,0 in the centre of the point cloud.
Thank you very much in advance for any hint!

```
parsed: format binary_little_endian 1.0
parsed: comment Created by CloudCompare v2.11 alpha (Anoia)
parsed: comment Created 30/04/2020 13:47
unknown header item: obj_info Generated by CloudCompare!
not implemented. contact martin@rapidlasso.com
parsed: obj_info Generated by CloudCompare!
parsed: element vertex 161617
parsed: property float x
parsed: property float y
parsed: property float z
parsed: property uchar red
parsed: property uchar green
parsed: property uchar blue
parsed: property float nx
parsed: property float ny
parsed: property float nz
Error in C_reader(ifiles, ofile, select, filter, filter_wkt) : 
  c++ exception (unknown reason)```

rapidlasso, 2020-05-02:
Can you post a link (!!!) to a sample PLY file?

spono, 2020-05-06:
sent by email

rapidlasso, 2020-05-13:
Fixed in the lastest version 200509.

closed 2020-05-13 by rapidlasso