LAStools/LAStools #98: lasnoise64 removing intensity field

smcavoy12, 2020-09-12:
Hello,

I've been happily using your noise removal tool, but found that while RGB color is preserved in the output, Intensity values are removed. I've been going over the readme and don't see anything related to this, is there an argument required keep the Intensity?

Here's the command I've been using:
`lasnoise64 -v -i Maeshowe.las -remove_noise -step 0.1`

The file processed with cloudcompare's noise filter (preserves intensity) 
https://pointcloud.ucsd.edu/Maeshowe.html

Processed with lasnoise64 (Intensity field is gone), I've checked to make sure the scalar wasn't just renamed, it's totally gone.
https://pointcloud.ucsd.edu/Maeshowe_lasnoise.html

rapidlasso, 2020-09-12:
Do you have a license to LAStools? This sounds like a typical "restricted use" issue. But it should be reported on the command line.

smcavoy12, 2020-09-12:
Ah! I see, yes. Iintensity, gps_time, user_data, and point_source_ID-zeroed" .

Thanks Martin, I'll buy a license.

closed 2020-09-12 by smcavoy12