LAStools/LAStools #103: Multi-character character constants are UB

abellgithub, 2020-12-31:
There are multiple instances of multi-character character constants in the code, like:

> [60/73] Building CXX object src/CMakeFiles/lasinfo.dir/lasinfo.cpp.o
../src/lasinfo.cpp:314:25: warning: illegal character encoding in character literal [-Winvalid-source-encoding]
      if (argv[i][0] == '<96>') argv[i][0] = '-';

This is not well-defined C++ code.

rapidlasso, 2021-01-01:
I am not entirely sure what you mean. Sometimes this character <96> shows up instead of a hyphen when people copy and paste command lines and that fixed the issue.

abellgithub, 2021-01-02:
'<96>' is not a well-defined character, as the warning shows. See

https://en.cppreference.com/w/cpp/language/character_literal

Case 6 applies, as does the note.

elfprince13, 2022-04-12:
```console
/home/ubuntu/LAStools/src/lasindex.cpp:144:25: error: character too large for enclosing character literal type
      if (argv[i][0] == '�') argv[i][0] = '-';
```