LAStools/LAStools #108: How install LAStool in Ubuntu ?

ambarja, 2021-04-14:
Hi, I would like use LAStool on ubuntu , but I don't know which are the steps that following to install, someone can I help me ? 
Thanks

closed 2021-04-19 by ambarja

jhx646018057, 2021-07-21:
Hello, did you install LAStools successfully?
I found the commands to install LAStools are as following:
"cd LAStools; mkdir build; cd build
cmake ..
sudo make install"
but when I code "sudo make install", there is :
"CMake Error at LASlib/src/cmake_install.cmake:141 (file):
  file INSTALL cannot find
  "/home/jhx/Downloads/LAStools(1)/LAStools/LASlib/src/laslib-config.cmake".
Call Stack (most recent call first):
  cmake_install.cmake:42 (include)

Makefile:73: recipe for target 'install' failed
make: *** [install] Error 1
"

fcakyon, 2021-09-09:
@jhx646018057 having the same error, have found the solution?

rhurlin, 2022-01-09:
It is exactly the same error, when I try to install v2.0.0 with cmake on FreeBSD. 

Compiler is Clang 13.0.0, cmake 3.22.1 with ninja 1.10.2.

BB-Heidelberg, 2022-01-10:
You have to use cmake.
Within the base dir just do
cmake .
This will update your build files according to your local setting.
May some warnings occur.
If errors occur check your cmake, gcc, linux,... setup.
Then run
cmake --build .
Everything will be compiled.
Good luck!

BB-Heidelberg, 2022-01-10:
According to your first comment you did "cmake .." but you should do "cmake ." - sometime a cmake setup use a ./build dir. but not here.

Bernd-Vogelgesang, 2022-11-12:
Hi there,
though I am using Linux for over 10 years, I am still lost with installing software this way. Please, is anybody able to give a full set of instructions how to install LASTools on my Linux Mint?

The README.txt states:

> _"... For Linux and MacOS
> the makefiles are included. Simply go into the root directory and run 'make':
> 
> unzip LAStools.zip
> cd LAStools/
> make
> 
> The compiled binary executables are or will be in the ./LAStools/bin directory. "

Doing this, stuff happens. But what next? The expected LASTools-commands are not available in the terminal.

Also tried variations of this cmake . -stuff, resulting in the same errors posted above.

I feel a bit silly. Thanx for any  help.

CharlesGaydon, 2022-11-21:
Using the officiel link here worked for me : https://downloads.rapidlasso.de/LAStools.zip
Then:
```bash
cf ~/
unzip LAStools.zip
cd LAStools/
make
```
Then use wanted library (e.g. lasinfo) with 
```bash
~/LAStools/bin/lasinfo
```