LAStools/LAStools #114: How to only read the "z" of ".laz" file ？

Lachiven, 2021-08-09:
I use decompress_selective but the read time have no chage at all. The I try to use the layer func of lastools, but dont konw how to put the attributes I want to the right layer.

kannes, 2022-01-07:
You got to provide some more information. What exactly do you mean by "only read the 'z'"? What exactly does your code or commandline invocations look like?

esilvia, 2022-01-13:
The lastools/laszip library functions and indeed the LAS format itself are written such that each point's data is included as discrete packets with every attribute, rather than storing each attribute for all points.

That is, you can't read a point's Z value without also reading its X, Y, Intensity, etc values. As you've observed, there would be no performance advantage to doing so because of the way that the LAS file itself is structured. See the current version of the specification for more information: https://www.asprs.org/divisions-committees/lidar-division/laser-las-file-format-exchange-activities

LAZ is simply a compressed version of LAS, so it follows many of the same rules.

This is in contrast to some other formats such as a database that might store all points' X values, then all points' Y values, then all points' Z values, etc.

rapidlasso, 2022-05-17:
Thanks Evon for that detailed explanation.

closed 2022-05-17 by rapidlasso