LAStools/LAStools #116: Error building make[2] on Linux

bwilder95, 2021-09-11:
Hello, I am trying to set up LAStools on Fedora 34 Linux and was unsuccessful in in building with the `make` command. Am I missing any steps in the installation process or is this an issue?

```
[brent@fedora LAStools]$ make
cd LASlib && make
make[1]: Entering directory '/home/brent/Downloads/LAStools/LASlib'
cd src && make
make[2]: Entering directory '/home/brent/Downloads/LAStools/LASlib/src'
g++  -c -O3 -Wall -Wno-deprecated -Wno-strict-aliasing -Wno-write-strings -Wno-unused-result -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -std=c++11 -I/usr/include/ -I../../LASzip/src -I../inc -I. ../../LASzip/src/laszip.cpp -o ../../LASzip/src/laszip.o
make[2]: g++: No such file or directory
make[2]: *** [Makefile:22: ../../LASzip/src/laszip.o] Error 127
make[2]: Leaving directory '/home/brent/Downloads/LAStools/LASlib/src'
make[1]: *** [Makefile:2: all] Error 2
make[1]: Leaving directory '/home/brent/Downloads/LAStools/LASlib'
make: *** [Makefile:2: all] Error 2

```

Best,
Brent

galitus, 2021-09-24:
> make[2]: g++: No such file or directory

I would say, you are missing g++, which is part of gcc.

try "which g++" and when there is no such file, install gcc for fedora

bwilder95, 2021-09-24:
Thanks @galitus this solved my problem

closed 2021-09-29 by bwilder95