LAStools/LAStools #117: AStools input E57 file as URL from Azure Blob file not working

maramusag, 2021-09-18:
For the input e57 file, it is stored on Azure Blob.
The URL is of the form
**https://<storageaccount>.blob.core.windows.net/<foldername>/<e57filename.e57>**
so might be:
**https://mystorage.blob.core.windows.net/workstuff/mye57.e57**

If you put this URL into your browser the file does download
however, the exe output say ERROR: file xxxx does not exist

and ideally would like to use it with the security access token on the end too, such as

**sp=r&st=2021-08-17T04:26:36Z&se=2021-09-18G11:26:36Z&spr=https&sv=2020-07-04&sr=b&sig=CejuKob6wGk0v5UNuf0Pwc%2BtQjFllxA4qAFh3%2FqFxAs%3D**

Is there any way to get e572las.exe to read this URL?
Escape characters?
I have tried lots of different URL encoding  and escape characters to no avail.

Jean-Romain, 2021-09-18:
LAstools read files on disk but doe not download files

zwnk, 2021-11-11:
you could use powershell and download the files by using `wget` aka `Invoke-WebRequest` https://mystorage.blob.core.windows.net/workstuff/mye57.e57 and after wards call e572las.exe on the downloaded file.

closed 2022-05-17 by rapidlasso