LAStools/LAStools #118: use GitHub releases to help package maintainers?

StephenWithPH, 2021-12-17:
I packaged LAStools for [Nix](https://search.nixos.org/packages?channel=21.11&show=LAStools&from=0&size=50&sort=relevance&type=packages&query=lastools). One of the challenges in keeping that up-to-date is the lack of machine-readable release metadata.

Due to that lack of release metadata, LAStools is far behind in most package repositories (see below from [repology](https://repology.org/project/lastools/information)):

[image]

Were LAStools to use GitHub's [releases](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases) function, that would enable some automation around keeping LAStools up-to-date where it's packaged.

Is this something you would consider?

rapidlasso, 2021-12-17:
Thanks for this hint, Stephen. Yes, we will consider this on next update.

esilvia, 2021-12-17:
I'll second this. lastools has always been one of the more difficult dependencies to maintain. It was so much work to compile with my VS compiler that I still just merge the changes into my personal fork.

StephenWithPH, 2022-01-26:
I see the 2.0 release on GitHub. Thank you very much!

closed 2022-01-26 by StephenWithPH

StephenWithPH, 2022-01-27:
I opened https://github.com/repology/repology-rules/pull/589 to clean up the repology ignore rule.