LAStools/LAStools #119: Problem about reading RGB from .las file

Philip035, 2022-01-07:
`		LASpoint& Point = lasreader->point;
		cout << Point.get_x() << endl;
		cout << Point.rgb[2] << endl;
`

There are at least 7 properties in my data: x, y, z, r, g, b, intensity.
As we all know, the rgb is ranged from 0 to 255, however the rgb information read by Lastools is over 255.
Is there any problem with my code?

esilvia, 2022-01-13:
RGB in LAS is encoded as a 16-bit integer ranging 0 to 65535 to allow greater bit depth and dynamic range. There is nothing wrong with your code.

Learn more by reading about point formats 2, 3, 5, 7, 8, and 10 in the LAS specification:
https://www.asprs.org/divisions-committees/lidar-division/laser-las-file-format-exchange-activities

Philip035, 2022-01-14:
> 

Thank you for providing the requested information!

Philip035, 2022-01-14:
> RGB (and intensity) in LAS is encoded as a 16-bit integer ranging 0 to 65535 to allow greater bit depth and dynamic range. There is nothing wrong with your code.
> 
> Learn more by reading about point formats 2, 3, 5, 7, 8, and 10 in the LAS specification: https://www.asprs.org/divisions-committees/lidar-division/laser-las-file-format-exchange-activities

Thank you for providing the requested information!

closed 2022-05-11 by Philip035