LAStools/LAStools #120: Executables aren't installed

yurivict, 2022-01-09:
Version 2.0

BB-Heidelberg, 2022-01-10:
Please precise! There is nothing to install. Unzip the zip. Run the tools in the .\bin directory.

yurivict, 2022-01-10:
It builds these executables and doesn't install them:
```lasindex las2txt las2las txt2las lasdiff lasinfo laszip lasmerge lasprecision```

BB-Heidelberg, 2022-01-10:
Perfect. If this executables are build just use them. See the *_README.txt file in the bin dir if you need further information about the tools.

yurivict, 2022-01-10:
In order to use them they need to be installed.

kannes, 2022-01-20:
Says who and why?

yurivict, 2022-01-21:
>> In order to use them they need to be installed.

> Says who and why?

Binaries are built, then the project is installed, packaged, distributed.
Binaries aren't in the package. It's kind of hard to use something that doesn't exist on the system.

kannes, 2022-01-21:
Which package are you referring to? Which/what "system"?

The package LASTools.zip from https://github.com/LAStools/LAStools/releases/tag/v2.0.0 does include the binaries.

yurivict, 2022-01-21:
[Here](https://cgit.freebsd.org/ports/tree/misc/lastools/Makefile) executables have to be installed manually because the project doesn't install them.

kannes, 2022-01-21:
I am sorry but you are being very difficult to help. Please write a detailed description what you are using, what you are doing and what you want the result to be. Otherwise this is a pointless waste of time and probably not a bug of LASTools at all.

yurivict, 2022-01-21:
I maintain the FreeBSD port linked above.
The port builds the package that is missing executables.
Instructions to install executables have to be added to the port manually.
The project should install them by itself.

esilvia, 2022-01-21:
Sounds like a cmake thing: https://cmake.org/cmake/help/book/mastering-cmake/chapter/Install.html

i.e., he's expecting the Makefile for lastools or laslib to do... something when he invokes `make install`. The only thing I see in the script linked by @yurivict is removing the "64" suffix from some of the executables.

yurivict, 2022-01-21:
Sorry, my bad.

The executables are actually installed by the project.

Sorry about the confusion.

closed 2022-01-21 by yurivict

kannes, 2022-01-21:
Thank you for maintaining that port and allowing others to benefit from your packaging!

yurivict, 2022-01-21:
You are welcome!

Thank you for developing the project!

kannes, 2022-01-22:
Oh I am just a user who sometimes tries to help triage bug reports here and there :)