LAStools/LAStools #121: Cloud optimized point cloud support

wiesehahn, 2022-01-26:
Hi, 
As far as I know the handling of cloud optimized point cloud data ([copc.io]()) is not yet support by LAStools. 
Are there any plans for this?

BB-Heidelberg, 2022-01-26:
As soon there is a "strong" request we will implement this. You are the first so far :)

Jean-Romain, 2022-01-26:
What do you mean by _strong_? I'm supporting this request. COPC is interesting and can become a standard but only if the biggest players provide a support for it.

wiesehahn, 2022-08-03:
I still think it would be a good addition, as the format is gaining more traction (e.g. COPC support in recent QGIS version).

lgouzin, 2024-04-03:
I randomly found it, it is now in Lastools. Works much quicker than importing a LAS straight to QGIS.
https://lastools.github.io/download/lascopcindex_README.md

wiesehahn, 2024-04-03:
thanks to @Jean-Romain

closed 2024-09-16 by rapidlasso