LAStools/LAStools #123: Read LAS (or LAZ) from memory location rather than files

MarkSMillman, 2022-03-07:
For a variety of reasons, I'd like to be able to read and write LAS and LAZ files from memory rather than files.  I have my own set of tools for specific processing tasks that work against mapped memory LAS files.  It would be efficient (i.e. minimize IO) if I could work with shared memory instances of my LAZ and LAS files rather than physical files.  Specifically, open a LAZ file as a mapped memory file and convert it to a shared memory instance of a LAS file, update the shared memory LAS file and then convert it back to a new LAZ file.  Reviewing the source code I can see bytestreamin_array.hpp and bytestreamout_array.hpp that look like they might do what I need.  But, I was wondering if anyone else has attempted to do this and could give me some pointers on how to configure LASreadOpener, LASreader, LASwriteOpener and LASwriter to use ByteStreamInArray and ByteStreamOutArray as their ByteStreams.

Thanks, Mark

MarkSMillman, 2022-03-08:
SOLVED -- My mistake was using the LASreadOpener.  I changed my code to use LASreaderLAS(ByteStreamInArray) and LASwriterLAS(ByteStreamOutArray) and created a new ByteStream[In/Out]MappedMemory classes that extend ByteStream[In/Out]ArrayLE which I construct by passing the MappedMemory object.  It works exactly as I'd hoped.

closed 2022-03-08 by MarkSMillman

Jean-Romain, 2022-03-09:
Could you share an example of that?

MarkSMillman, 2022-03-09:
Sure, give me a day or two though.

MarkSMillman, 2022-03-10:
Unfortunately, this is not a stand-alone example.  It uses a MappedFile class that I use to wrap boost::iostream::mapped_file.  But, for the purposes of the example it could be replaced with heap memory or shared memory.  This is just prototype code so I make no promises but it should give you an idea of  how to build your own.

[LASLibTestMark.cpp.txt](https://github.com/LAStools/LAStools/files/8219413/LASLibTestMark.cpp.txt)

MarkSMillman, 2022-04-15:
I found a memory leak in my example code.  The ByteStreamOutMFile constructor should pass 0 to the base class.
`		ByteStreamOutMFile(_MappedBase* mappedMemory) : ByteStreamOutArrayLE(0) {
			if (data != nullptr)
				free(data);
			data = (U8*)mappedMemory->Data();
			alloc = (I64)mappedMemory->Size();
			size = 0;
			curr = 0;
		}`
Because the default alloc is 4096, by passing a zero the base class calls malloc(0) which does not allocate memory but (with Visual Studio) returns a pseudo pointer which can be passed to free.