LAStools/LAStools #130: error: character too large for enclosing character literal type

yurivict, 2022-06-19:
```
/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/lasindex.cpp:144:25: error: character too large for enclosing character literal type
      if (argv[i][0] == '�') argv[i][0] = '-';
                        ^
1 error generated.
```

Version: 2.0.1
clang-13
FreeBSD 13.1

rapidlasso, 2022-06-19:
same as #103 - if you or your compiler dont like this just remove the character with '-' or remove the statement

closed 2022-06-19 by rapidlasso

yurivict, 2022-06-19:
This is a C++ bug - an invalid C++ code. This isn't a matter of liking this.

fmvaldezg, 2022-10-11:
warning: illegal character encoding in character literal [-Winvalid-source-encoding]
      if (argv[i][0] == '<96>') argv[i][0] = '-';

fmvaldezg, 2022-10-11:
error: character too large for enclosing character literal type
      if (argv[i][0] == '�') argv[i][0] = '-';

rapidlasso, 2022-12-15:
This is not a bug.
The code is
      if (argv[i][0] == '–') argv[i][0] = '-';
If you have a look with a hex editor you see the first hyphen as 0x96, the second one as 0x2D.
The file is plain ascii.
The recomended codepage to show this in a nice way is [1252](https://en.wikipedia.org/wiki/Windows-1252).
But, of course, the compiler does not take care about the codespace, its just a char...
If your editor displays something funky like "�": He try to read the file as an utf-8 file, which is wrong.
If you editor display <96> he just does not use the expected codepage.
If your compiler produce an error or warning you may just saved the crop produced by you editor.

yurivict, 2022-12-15:
C++ doesn't allow any characters from the extended ASCII table (single byte characters with values >=0x80).

You needs to simply replace it with 0x96.

rapidlasso, 2023-02-02:
Done with the commit today