LAStools/LAStools #135: Multicore support in linux (Ubuntu)

jfprieur, 2022-09-02:
Hello, I am trying to migrate our ALS workflow to the cloud, I am having issues getting multicore support to run under linux instances for laszip which is the first part of our process. I followed the instructions to compile from source, I can succesfully execute laszip on a single core so I believe it install the binary properly.

When I try to use the -cores option I get 

`WARNING: not compiled with multi-core batching. ignoring '-cores' ...`

I found some older messages on the Google Group (is that the better forum to ask questions in than this one?) that seem to hint that it was never implemented and some workarounds that unfortunately do not work for me either.

Is multicore under linux implemented for laszip at this moment?

Jean-Romain, 2022-09-02:
**Disclaimer: I do not pretend to give a definitive answer. I'm not the owner but I read the source code.**

The source code of open-source binaries (`laszip`, `lasindex` and co) contains multiple occurances of preprocessor conditional statements like 

```r
#ifdef COMPILE_WITH_MULTI_CORE
extern int laszip_multi_core(int argc, char *argv[], GeoProjectionConverter* geoprojectionconverter, LASreadOpener* lasreadopener, LASwriteOpener* laswriteopener, int cores, BOOL cpu64);
#endif
```

So, in theory, you add the flag `-DCOMPILE_WITH_MULTI_CORE` in the `makefile` and it should compile with multicore.

```
COPTS     = -O3 -Wall -Wno-deprecated -Wno-strict-aliasing -Wno-write-strings -Wno-unused-result -DNDEBUG -DCOMPILE_WITH_MULTI_CORE
```

However there is not a single c++ file defining `laszip_multi_core` and other similar multi-core functions. Consequently if you compile your binaries with multicore if fails during the linking step with an  expected error

```
laszip.cpp:(.text.startup+0x94f) : undefined reference to « laszip_multi_core(int, char**, GeoProjectionConverter*, LASreadOpener*, LASwriteOpener*, int, bool) »
```

For `laszip`, `las2las`, `lasindex` and other open source binaries, if the multicore support exists, it is not part of the open source code available in this repo.

rapidlasso, 2022-09-02:
As owner of the software: Thanks @Jean-Romain, the answer is correct.
There is no use of the multi-core option in the open source part - this compilerswitch is just for the commercial part of LAStools.

jfprieur, 2022-09-02:
Thank you (merci!) to JRR for the quick answer and rapidlasso for the quick confirmation! I will explore the workarounds.

closed 2022-09-02 by jfprieur