LAStools/LAStools #136: error: character too large for enclosing character literal type

yurivict, 2022-10-11:
2.0.1 fails:
```
[ 85% 6/7] /usr/local/libexec/ccache/c++ -DNDEBUG -D_CRT_SECURE_NO_WARNINGS -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/../LASzip/src -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/../LASlib/inc -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/LASlib/inc -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/LASzip/src -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -MD -MT src/CMakeFiles/lasindex.dir/lasindex.cpp.o -MF src/CMakeFiles/lasindex.dir/lasindex.cpp.o.d -o src/CMakeFiles/lasindex.dir/lasindex.cpp.o -c /disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/lasindex.cpp
FAILED: src/CMakeFiles/lasindex.dir/lasindex.cpp.o 
/usr/local/libexec/ccache/c++ -DNDEBUG -D_CRT_SECURE_NO_WARNINGS -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/../LASzip/src -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/../LASlib/inc -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/LASlib/inc -I/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/LASzip/src -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer -MD -MT src/CMakeFiles/lasindex.dir/lasindex.cpp.o -MF src/CMakeFiles/lasindex.dir/lasindex.cpp.o.d -o src/CMakeFiles/lasindex.dir/lasindex.cpp.o -c /disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/lasindex.cpp
/disk-samsung/freebsd-ports/misc/lastools/work/LAStools-2.0.1/src/lasindex.cpp:144:25: error: character too large for enclosing character literal type
      if (argv[i][0] == '�') argv[i][0] = '-';
                        ^
1 error generated.
```

clang-14
OS: FreeBSD 13.1 STABLE

yurivict, 2022-10-11:
Duplicate of https://github.com/LAStools/LAStools/issues/130

rapidlasso, 2022-12-15:
This is not a bug.
The code is
if (argv[i][0] == '–') argv[i][0] = '-';
If you have a look with a hex editor you see the first hyphen as 0x96, the second one as 0x2D.
The file is plain ascii.
The recomended codepage to show this in a nice way is [1252](https://en.wikipedia.org/wiki/Windows-1252).
But, of course, the compiler does not take care about the codepage, its just a char...
If your editor displays something funky like "�": He try to read the file as an utf-8 file, which is wrong.
If you editor display <96> he just does not use the expected codepage.
If your compiler produce an error or warning you may just saved the crop produced by you editor.

closed 2022-12-15 by rapidlasso