LAStools/LAStools #139: Scaling values for -scale_n optoins in las2las are being truncated.

lenusmars, 2022-11-28:
As of Version 220801

I often deliver two copies of a dataset to clients.  One is normally in state plane coordinates and the other is a set where the coordinate values have been multiplied by a combined factor (CF) to bring them closer to coordinates on the ground. using the scale_x and scale_y commands. (Z is not scaled.)

The standard combined factor is 8 decimal places.  (ex. 1.00008780) I have discovered that las2las truncates the factor to 6 decimal places 1.000088 when scaling the coordinates which leads to inaccuracies when matching to vector data collected against the las scaled by other tools.

I'm hoping the command can be fixed to respect the scaling factor to 8 decimal places (Common in survey work.)

BB-Heidelberg, 2022-12-02:
I would rather crop the other data after some decimals. In terms of precision it does not make sense and may blow up your data. Please see [this blog entry](https://rapidlasso.com/2019/05/06/how-many-decimal-digits-for-storing-longitude-latitude/).
Also it does not make much sense to close and reopen a issue to get my attention - ok, this time it worked.

lenusmars, 2022-12-15:
> 

The close and open was an accident.  I apologize for that. 

I don’t think you are understanding me.  I am not talking about the scaling value that controls where the decimal point is placed on the coordinates stored as integers.  I’m not referring to storage.

The -scale_x and -scale_y command  are not about how many decimal places the coordinates are stored with.  They multiply  the coordinates by a scaling value.  This is commonly done in the US in state surveys.   The projected coordinates (usually some form of United States State Plane in US Survey Feet) are brought closer to the correct coordinates on the ground.  The original coordinates are figured relative to the ellipsoid.   The Combined factor used to scale with brings the x y coordinate values closer to actual ground level.

I repeat, this is not about ‘scaling’ in terms of how many decimal points to store on a coordinate.  I’m  aware of that and those accuracy levels.  This issue that the program is  truncating the number I’m entering to spread out the xy coordinates, and therefore putting the final coordinates in the wrong place.  When the scale factor is applied and the center of the system is 1,000,0000 units away, it is an issue.  

All I’m asking is that it does not truncate the number I’m inputting for this purpose.