LAStools/LAStools #141: las2las updating WKT projection results in double horizontal

brettedw, 2022-12-15:
Using the command: 
`las2las -cpu64 -i Sample_GUID_ForNV5G_epsg3005_1.las
	     -set_ogc_wkt -epsg 3153 -vertical_cgvd2013`

Results in a VLR that looks like this:
<pre>variable length header record 2 of 2:
  reserved             0
  user ID              'LASF_Projection'
  record ID            2112
  length after header  972
  description          'by LAStools of rapidlasso GmbH'
    WKT OGC COORDINATE SYSTEM:
    COMPD_CS["<b>NAD83(CSRS) / NAD83(CSRS)</b> / BC Albers + CGVD2013",PROJCS["<b>NAD83(CSRS) / NAD83(CSRS)</b> / BC Albers",GEOGCS["NAD83(CSRS)",DATUM["NAD83_Canadian_Spatial_Reference_System",SPHEROID["GRS 1980",6378137,298.257222101,AUTHORITY["EPSG","7019"]],AUTHORITY["EPSG","6140"]],PRIMEM["Greenwich",0,AUTHORITY["EPSG","8901"]],UNIT["degree",0.01745329251994328,AUTHORITY["EPSG","9122"]],AUTHORITY["EPSG","4617"]],PROJECTION["Albers_Conic_Equal_Area"],PARAMETER["standard_parallel_1",50],PARAMETER["standard_parallel_2",58.5],PARAMETER["latitude_of_center",45],PARAMETER["longitude_of_center",-126],PARAMETER["false_easting",1000000],PARAMETER["false_northing",0],UNIT["metre",1,AUTHORITY["EPSG","9001"]],AXIS["Easting",EAST],AXIS["Northing",NORTH],AUTHORITY["EPSG","3153"]],VERT_CS["CGVD2013",VERT_DATUM["Canadian Geodetic Vertical Datum of 2013",2005,AUTHORITY["EPSG","1127"]],UNIT["metre",1.0,AUTHORITY["EPSG","9001"]],AXIS["Gravity-related height",UP],AUTHORITY["EPSG","6647"]]]</pre>

As you can see it writes a double horizontal datum. We've tried a few ways at getting around this without a whole lot of success

rsking, 2022-12-15:
I've had a look at the WKT, and it all seems fine to me... what do you think is wrong?

Formatted WKT
```
COMPD_CS[
	"NAD83(CSRS) / NAD83(CSRS) / BC Albers + CGVD2013",
	PROJCS[
		"NAD83(CSRS) / NAD83(CSRS) / BC Albers",
		GEOGCS[
			"NAD83(CSRS)",DATUM[
				"NAD83_Canadian_Spatial_Reference_System",
				SPHEROID[
					"GRS 1980",
					6378137,
					298.257222101,
					AUTHORITY[
						"EPSG",
						"7019"]],
				AUTHORITY[
					"EPSG",
					"6140"]],
			PRIMEM[
				"Greenwich",
				0,
				AUTHORITY[
					"EPSG",
					"8901"]],
			UNIT[
				"degree",
				0.01745329251994328,
				AUTHORITY[
					"EPSG",
					"9122"]],
			AUTHORITY[
				"EPSG",
				"4617"]],
		PROJECTION["Albers_Conic_Equal_Area"],
		PARAMETER[
			"standard_parallel_1",
			50],
		PARAMETER[
			"standard_parallel_2",
			58.5],
		PARAMETER[
			"latitude_of_center",
			45],
		PARAMETER[
			"longitude_of_center",
			-126],
		PARAMETER[
			"false_easting",
			1000000],
		PARAMETER[
			"false_northing",
			0],
		UNIT[
			"metre",
			1,
			AUTHORITY[
				"EPSG",
				"9001"]],
		AXIS[
			"Easting",
			EAST],
		AXIS[
			"Northing",
			NORTH],
		AUTHORITY[
			"EPSG",
			"3153"]],
	VERT_CS[
		"CGVD2013",
		VERT_DATUM[
			"Canadian Geodetic Vertical Datum of 2013",
			2005,
			AUTHORITY["EPSG","1127"]],
		UNIT[
			"metre",
			1.0,
			AUTHORITY["EPSG","9001"]],
		AXIS[
			"Gravity-related height",
			UP],
		AUTHORITY[
			"EPSG",
			"6647"]]]
```

brettedw, 2022-12-15:
NAD83(CSRS) should not be written twice back to back in the two bolded locations. It should look like this: 

```
> COMPD_CS[
> 	"NAD83(CSRS) / BC Albers + CGVD2013",
> 	PROJCS[
> 		"NAD83(CSRS) / BC Albers",
> 		GEOGCS[
> 			"NAD83(CSRS)",DATUM[
> 				"NAD83_Canadian_Spatial_Reference_System",
> 				SPHEROID[
> 					"GRS 1980",
> 					6378137,
> 					298.257222101,
> 					AUTHORITY[
> 						"EPSG",
> 						"7019"]],
> 				AUTHORITY[
> 					"EPSG",
> 					"6140"]],
> 			PRIMEM[
> 				"Greenwich",
> 				0,
> 				AUTHORITY[
> 					"EPSG",
> 					"8901"]],
> 			UNIT[
> 				"degree",
> 				0.01745329251994328,
> 				AUTHORITY[
> 					"EPSG",
> 					"9122"]],
> 			AUTHORITY[
> 				"EPSG",
> 				"4617"]],
> 		PROJECTION["Albers_Conic_Equal_Area"],
> 		PARAMETER[
> 			"standard_parallel_1",
> 			50],
> 		PARAMETER[
> 			"standard_parallel_2",
> 			58.5],
> 		PARAMETER[
> 			"latitude_of_center",
> 			45],
> 		PARAMETER[
> 			"longitude_of_center",
> 			-126],
> 		PARAMETER[
> 			"false_easting",
> 			1000000],
> 		PARAMETER[
> 			"false_northing",
> 			0],
> 		UNIT[
> 			"metre",
> 			1,
> 			AUTHORITY[
> 				"EPSG",
> 				"9001"]],
> 		AXIS[
> 			"Easting",
> 			EAST],
> 		AXIS[
> 			"Northing",
> 			NORTH],
> 		AUTHORITY[
> 			"EPSG",
> 			"3153"]],
> 	VERT_CS[
> 		"CGVD2013",
> 		VERT_DATUM[
> 			"Canadian Geodetic Vertical Datum of 2013",
> 			2005,
> 			AUTHORITY["EPSG","1127"]],
> 		UNIT[
> 			"metre",
> 			1.0,
> 			AUTHORITY["EPSG","9001"]],
> 		AXIS[
> 			"Gravity-related height",
> 			UP],
> 		AUTHORITY[
> 			"EPSG",
> 			"6647"]]]
```

rsking, 2022-12-15:
Yep, fair enough. `gdalsrsinfo EPSG:3153 -o wkt1` definitely produces the WKT1 without the extra `NAD83(CSRS)` in the name.