LAStools/LAStools #145: License question

Canivek, 2023-02-28:
Hello,
I would like to read/write las/laz files on iOS. Is this allowed to statically link with this library?

I have found this old GitHub issue #6 that links to https://groups.google.com/g/lastools/c/Bs9YBXzXXcg saying:

> In order to allow LASzip to be used on mobile devices and operating systems that do not allow dynamic linking we have always given those who asked permission to link statically provided they did not alter the format. Recently we have officially added this "static linking exception" to the LASzip library.
> 
> http://github.com/LAStools/LAStools/blob/master/LASzip/LICENSE.txt

But the link leads to a 404, and I can't find a mention of static linking exception in the current state of the repository.

rapidlasso, 2023-02-28:
To read/write LAS/LAZ you can use 
https://github.com/LASzip/LASzip
laszip is a part/subset of lastools and "Apache Public License 2.0", so you can static link this library.

Canivek, 2023-03-02:
Oh, completely missed that repository. That's perfect, thanks.

closed 2023-03-02 by Canivek