LAStools/LAStools #148: undefined reference to `LASreadItemCompressed_BYTE14_v3::LASreadItemCompressed_BYTE14_v3

fspegni, 2023-03-28:
I just downloaded the git repository in Ubuntu 22.04.2 LTS (kernel Linux 5.19.0-38-generic #39~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 21:16:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux). 
Following the README, I just `cd`-ed in the repository folder and typed `make`. I get the following errors (after several warnings)
[output.txt](https://github.com/LAStools/LAStools/files/11087722/output.txt)
:

```
/usr/bin/ld: ../lib/liblas.a(lasreadpoint.o): in function `LASreadPoint::setup(unsigned int, LASitem const*, LASzip const*)':
lasreadpoint.cpp:(.text+0x3b6): undefined reference to `LASreadItemCompressed_BYTE14_v3::LASreadItemCompressed_BYTE14_v3(ArithmeticDecoder*, unsigned int, unsigned int)'
/usr/bin/ld: lasreadpoint.cpp:(.text+0x47c): undefined reference to `LASreadItemCompressed_WAVEPACKET14_v3::LASreadItemCompressed_WAVEPACKET14_v3(ArithmeticDecoder*, unsigned int)'
/usr/bin/ld: lasreadpoint.cpp:(.text+0x4b2): undefined reference to `LASreadItemCompressed_RGBNIR14_v3::LASreadItemCompressed_RGBNIR14_v3(ArithmeticDecoder*, unsigned int)'
/usr/bin/ld: lasreadpoint.cpp:(.text+0x51a): undefined reference to `LASreadItemCompressed_RGB14_v3::LASreadItemCompressed_RGB14_v3(ArithmeticDecoder*, unsigned int)'
/usr/bin/ld: lasreadpoint.cpp:(.text+0x552): undefined reference to `LASreadItemCompressed_POINT14_v3::LASreadItemCompressed_POINT14_v3(ArithmeticDecoder*, unsigned int)'
/usr/bin/ld: ../lib/liblas.a(laswritepoint.o): in function `LASwritePoint::setup(unsigned int, LASitem const*, LASzip const*)':
laswritepoint.cpp:(.text+0x360): undefined reference to `LASwriteItemCompressed_BYTE14_v3::LASwriteItemCompressed_BYTE14_v3(ArithmeticEncoder*, unsigned int)'
/usr/bin/ld: laswritepoint.cpp:(.text+0x3da): undefined reference to `LASwriteItemCompressed_WAVEPACKET14_v3::LASwriteItemCompressed_WAVEPACKET14_v3(ArithmeticEncoder*)'
/usr/bin/ld: laswritepoint.cpp:(.text+0x402): undefined reference to `LASwriteItemCompressed_RGBNIR14_v3::LASwriteItemCompressed_RGBNIR14_v3(ArithmeticEncoder*)'
/usr/bin/ld: laswritepoint.cpp:(.text+0x44a): undefined reference to `LASwriteItemCompressed_RGB14_v3::LASwriteItemCompressed_RGB14_v3(ArithmeticEncoder*)'
/usr/bin/ld: laswritepoint.cpp:(.text+0x472): undefined reference to `LASwriteItemCompressed_POINT14_v3::LASwriteItemCompressed_POINT14_v3(ArithmeticEncoder*)'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:19: lasexample] Error 1
make[2]: Leaving directory '/home/spegni/git/LAStools/LASlib/example'
make[1]: *** [Makefile:3: all] Error 2
make[1]: Leaving directory '/home/spegni/git/LAStools/LASlib'
make: *** [Makefile:2: all] Error 2
```

I attach the full compilation output, in case some information may be relevant. Do I miss any requirement?

Eastcow1990, 2023-07-07:
hi, is fixed?

Jean-Romain, 2023-07-07:
Which version of `g++` are you using?