LAStools/LAStools #151: while the LASpoint is different with the lasreaer->point after = operator?

daxue2017, 2023-09-16:
if use    reference below, 
LASpoint &point = lasreaer->point;
if use    =operator below, 

LASpoint point1 = lasreaer->point;

after test,   found the point is not equal to point1, 

so how to get a equal copy from the lasreaer->point?

Jean-Romain, 2023-09-16:
The copy operator of a `LASpoint` is also a conversion operator. If `point1` is initialized in another format it may be different from `point` after `=` operator.

Now without a reproducible example it is hard to say. What do you mean by different? Coordinates and attributes are expected to be the same. Pointer to `LASttributer`, `LASzip` or other internal pointers might be different.

daxue2017, 2023-09-17:
ok, may be the point1 dont init. it is the same  where the point1 init then use the =operator

closed 2023-09-20 by daxue2017