LAStools/LAStools #160: How to use e572las on unix machine (macos or linux)

Phong95, 2023-11-13:
I greatly appreciate your work on the 'e572las' tool; it's very useful. 
Is there any way to use that tool on Unix machines like macOS and Linux?

rapidlasso, 2023-11-15:
We just released our Linux binary for the E57 to LAS/LAZ converter.
See this [rapidlasso blog entry](https://rapidlasso.de/e57-to-las-converter-for-linux/).

closed 2023-11-15 by rapidlasso

Phong95, 2023-11-16:
Hi @rapidlasso,
Im using Mac mini M1 with MacOs Ventura 13.3.1, 
i download and extract binaries to folder path `/Users/phong/Downloads/e572las/`

After cd to that folder path, i followed the instructions correctly, and here is result:
`
phong@Mac-mini-cua-phong e572las % export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/Users/phong/Downloads/e572las
phong@Mac-mini-cua-phong e572las % ./e572las64 -version                                                  
zsh: exec format error: ./e572las64
`
Did i do something wrong or missing any step?