LAStools/LAStools #165: Compile Fails

shaunakv1, 2024-01-17:
Compile fails after following commit to master:

[6944b1a393e689f1c450519eb39c9a8090d2b775](https://github.com/LAStools/LAStools/tree/6944b1a393e689f1c450519eb39c9a8090d2b775)

Docker Image:   condaforge/mambaforge:latest
Digest: bd06dcc2d46c (linux/amd64)

Relevant Docker File:

``` Dockerfile

FROM condaforge/mambaforge:latest as build

ENV LANG=C.UTF-8 LC_ALL=C.UTF-8

RUN mamba install -c conda-forge -y gdal pdal nodejs gmt python-pdal && conda clean --all

RUN apt-get update && \
  DEBIAN_FRONTEND=noninteractive apt-get install -y python3-pip sqlite3 \
  zip unzip \
  vim \
  dos2unix openjdk-8-jre

# RUN apt-get update && apt-get install -y  
# install Saxon
ARG SAXON_VERSION=9.4
WORKDIR /usr/src/Saxon-HE
RUN curl https://repo1.maven.org/maven2/net/sf/saxon/Saxon-HE/${SAXON_VERSION}/Saxon-HE-${SAXON_VERSION}.jar -o Saxon-HE.jar

# install LAStools
RUN mkdir /usr/src/LAStools
WORKDIR /usr/src/LAStools
RUN curl -sL https://github.com/LAStools/LAStools/archive/refs/tags/v2.0.2.zip -o lastools.zip
RUN unzip lastools.zip 
WORKDIR /usr/src/LAStools/LAStools-2.0.2
RUN make && cp bin/las2las /usr/bin

```

Error:

```bash
#17 6.193 g++  -c -O3 -Wall -Wno-deprecated -Wno-strict-aliasing -Wno-write-strings -Wno-unused-result -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -std=c++11 -I/usr/include/ -I../../LASzip/src -I../inc -I. ../../LASzip/src/lasindex.cpp -o ../../LASzip/src/lasindex.o
#17 6.445 ../../LASzip/src/lasindex.cpp:63:9: error: ‘unordered_map’ does not name a type
#17 6.445    63 | typedef unordered_map<I32,U32> my_cell_hash;
#17 6.445       |         ^~~~~~~~~~~~~
#17 6.447 ../../LASzip/src/lasindex.cpp: In member function ‘void LASindex::complete(U32, I32, BOOL)’:
#17 6.447 ../../LASzip/src/lasindex.cpp:116:5: error: ‘my_cell_hash’ was not declared in this scope
#17 6.447   116 |     my_cell_hash cell_hash[2];
#17 6.447       |     ^~~~~~~~~~~~
#17 6.449 ../../LASzip/src/lasindex.cpp:121:7: error: ‘cell_hash’ was not declared in this scope
#17 6.449   121 |       cell_hash[hash1][interval->index] = interval->full;
#17 6.449       |       ^~~~~~~~~
#17 6.450 ../../LASzip/src/lasindex.cpp:123:12: error: ‘cell_hash’ was not declared in this scope
#17 6.450   123 |     while (cell_hash[hash1].size())
#17 6.450       |            ^~~~~~~~~
#17 6.450 ../../LASzip/src/lasindex.cpp:134:7: error: ‘my_cell_hash’ is not a class, namespace, or enumeration
#17 6.450   134 |       my_cell_hash::iterator hash_element_inner;
#17 6.450       |       ^~~~~~~~~~~~
#17 6.450 ../../LASzip/src/lasindex.cpp:135:7: error: ‘my_cell_hash’ is not a class, namespace, or enumeration
#17 6.450   135 |       my_cell_hash::iterator hash_element_outer = cell_hash[hash1].begin();
#17 6.450       |       ^~~~~~~~~~~~
#17 6.451 ../../LASzip/src/lasindex.cpp:136:14: error: ‘hash_element_outer’ was not declared in this scope
#17 6.451   136 |       while (hash_element_outer != cell_hash[hash1].end())
#17 6.451       |              ^~~~~~~~~~~~~~~~~~
#17 6.453 ../../LASzip/src/lasindex.cpp:148:17: error: ‘hash_element_inner’ was not declared in this scope
#17 6.453   148 |                 hash_element_inner = hash_element_outer;
#17 6.453       |                 ^~~~~~~~~~~~~~~~~~
#17 6.455 ../../LASzip/src/lasindex.cpp:152:17: error: ‘hash_element_inner’ was not declared in this scope
#17 6.455   152 |                 hash_element_inner = cell_hash[hash1].find(indices[i]);
#17 6.455       |                 ^~~~~~~~~~~~~~~~~~
#17 6.456 ../../LASzip/src/lasindex.cpp:154:19: error: ‘hash_element_inner’ was not declared in this scope
#17 6.456   154 |               if (hash_element_inner != cell_hash[hash1].end())
#17 6.456       |                   ^~~~~~~~~~~~~~~~~~
#17 6.495 make[2]: Leaving directory '/usr/src/LAStools/LAStools-master/LASlib/src'
#17 6.495 make[2]: *** [Makefile:22: ../../LASzip/src/lasindex.o] Error 1
#17 6.496 make[1]: *** [Makefile:2: all] Error 2
#17 6.497 make[1]: Leaving directory '/usr/src/LAStools/LAStools-master/LASlib'
#17 6.498 make: *** [Makefile:2: all] Error 2
#17 ERROR: process "/bin/sh -c make && cp bin/las2las /usr/bin" did not complete successfully: exit code: 2
------
 > [11/25] RUN make && cp bin/las2las /usr/bin:
6.455   152 |                 hash_element_inner = cell_hash[hash1].find(indices[i]);
6.455       |                 ^~~~~~~~~~~~~~~~~~
6.456 ../../LASzip/src/lasindex.cpp:154:19: error: ‘hash_element_inner’ was not declared in this scope
6.456   154 |               if (hash_element_inner != cell_hash[hash1].end())
6.456       |                   ^~~~~~~~~~~~~~~~~~
6.495 make[2]: Leaving directory '/usr/src/LAStools/LAStools-master/LASlib/src'
6.495 make[2]: *** [Makefile:22: ../../LASzip/src/lasindex.o] Error 1
6.496 make[1]: *** [Makefile:2: all] Error 2
6.497 make[1]: Leaving directory '/usr/src/LAStools/LAStools-master/LASlib'
6.498 make: *** [Makefile:2: all] Error 2
```

BB-Heidelberg, 2024-01-17:
fixed; should compile now at all targets

closed 2024-01-17 by BB-Heidelberg

shaunakv1, 2024-01-18:
@BB-Heidelberg  Awesome, thanks for a quick turnaround!