LAStools/LAStools #171: Improvements for programmtically catching laslib/laszip messages

ottointhesky, 2024-02-02:
Errors, warnings and messages in laslib/laszib libraries are currently always written to stderr.  For code that use the libraries directly it would be beneficial to be more flexible: 
* Catching and custom logging of messages is a hassle since it requires redirecting (and parsing) the stderr to a custom pipe
* When reading/writing multiple files with multiple threads the pipe catching work-a-round makes it impossible to relate the messages to the corresponding file

The solution to this issue is quite simple:  
Instead of directly writing to stderr an error handling concept with definable callbacks as e.g. in GDAL (https://gdal.org/doxygen/cpl__error_8h.html) could be used. There a thread dependent stack of callbacks can be defined at which only the last one on the stack gets called. This way different binaries running within one process that use the same laslib/laszip DLL will not interfere (if the callbacks are pushed and removed correctly).

Furthermore, I suggest to categorise the messages via an enum:  _error_, _warning_, _info_, _verbose_, … Currently, the messages are prefixed with ‘ERROR:’ and ‘WARNING:’ (Info messages at not prefixed). Using the enum the prefix can be removed, but still the exact same message can be reproduced, just in case an existing code/software rely on the current message format.

I’m willing to provide a pull requests (if the idea finds acceptance). Looking forward to your feedback…

BB-Heidelberg, 2024-02-02:
Thanks for the suggestion! Maybe we will do some improvements at the output system some day. Right now may users use the WARNING: and ERROR: stderr output for further processing, so in terms of backwards compatibility it is not the best idea to change this behavior.

closed 2024-02-02 by BB-Heidelberg

ottointhesky, 2024-02-02:
I totally understand that, which is why I would keep **all** message **exactly the same** (including the prexfix til the last character) in the default message output function. e.g.: 

`fprintf(stderr,"ERROR: file pointer is zero\n") `
change to
`LASMessage(LAS_ERROR, "file pointer is zero\n")`

LASMessage calls the actual message handling function which supports user defined message calllback. But per default the  LASDefaultMessage function is called:

```
void LASDefaultMessage(LASMessageType type, const char *text) 
{
   switch (type) { 
   case LAS_ERROR:
     fprintf(stderr,"ERROR: "+text);
     return;
   case LAS_WARNING:
     fprintf(stderr,"WARNING: "+text);
     return;
   default:
     fprintf(stderr,text);
   }
}
```

Hence, from outside there is NOT a single charater changed in the messages send to stderr, but code that uses the library directly can intercept the message very easy...

Please reconsider!