LAStools/LAStools #183: Inconsistent character encoding for opening files

SunBlack, 2024-03-13:
We recently stumbled across the fact that we cannot open files with special characters, such as German umlauts.

As we are using Qt our code was like this:

```cpp
QString relativeFilePath = QString::fromUtf8("las/pointcloud_äöü.las"); // Source file UTF-8 encoded
LASreadOpener lasreadopener;
lasreadopener.set_file_name(relativeFilePath.toLocal8Bit().constData());
auto lasReader = lasreadopener.open();
if (!lasReader)
{
	qWarning() << QObject::tr("Cannot open file %1.").arg(relativeFilePath);
	return;
}
```

This test case failed, so I debugged it. It failed here:
https://github.com/LAStools/LAStools/blob/c3d4641d35b1e26a6982542a8db366daeb9fa4cd/LASlib/src/lasreader_las.cpp#L57-L65

As you can see `file_name` will be converted from UTF-8 to UTF-16 there. So I adjusted my code to use `toUtf8()` instead of `toLocal8Bit()`. This sadly didn't fixed the issue, as now `set_file_name` is failing. Reason is here:
https://github.com/LAStools/LAStools/blob/c3d4641d35b1e26a6982542a8db366daeb9fa4cd/LASlib/src/lasreader.cpp#L2936-L2938

As you can see `FindFirstFile` is used. This method is defined in `fileapi.h` as:
```cpp
#ifdef UNICODE
#define FindFirstFile  FindFirstFileW
#else
#define FindFirstFile  FindFirstFileA
#endif // !UNICODE
```
As `UNICODE` is not defined (and I don't see any export in the CMake script for it), [`FindFirstFileA`](https://learn.microsoft.com/de-de/windows/win32/api/fileapi/nf-fileapi-findfirstfilea) will be used. As it has the suffix `A`, and not `W`, now, it expects a `LPCSTR`, which is a null-terminated string of 8-bit Windows (ANSI) characters. Remember: The `LASreaderLAS::open` called `UTF8toUTF16`, so this expects an UTF-8 string.

I'm not sure why `set_file_name` does more than just storing the given path, but the issue can currently worked around by not using `set_file_name`:
```cpp
LASreadOpener lasreadopener;
auto lasReader = lasreadopener.open(relativeFilePath.toLocal8Bit().constData());
```

BB-Heidelberg, 2024-03-13:
Not supporting UTF8 is a known issue for a long time. Because it is not a real improvement of LiDAR data processing it is quite low prioritized by our team. But if someone likes to fix this we will be glad about this, just post a pull request.

h-thiel, 2024-03-14:
> Not supporting UTF8 is a known issue for a long time.

My intention with the issue was not so much to address the issue with UTF-8, but rather to point out that it is inconsistent. If only ANSI was supported, it would be okay. The question for me was rather: Why does `set_file_name` do more than just storing the file name, which is then used by the `open` method later. Unfortunately, I can't find any documentation for the method that explains what the method is supposed to do, apart from saving.

SunBlack, 2024-03-14:
> Not supporting UTF8 is a known issue for a long time.

My intention with the issue was not so much to address the issue with UTF-8, but rather to point out that it is inconsistent. If only ANSI was supported, it would be okay. The question for me was rather: Why does set_file_name do more than just storing the file name, which is then used by the open method later. Unfortunately, I can't find any documentation for the method that explains what the method is supposed to do, apart from saving.

rapidlasso, 2024-06-24:
done with #197, not full utf-8 support right now, but full ANSI support

closed 2024-06-24 by rapidlasso