LAStools/LAStools #187: Tagged open source release?

hobu, 2024-04-15:
It has been almost 1.5 years since the last tagged open source release (see https://github.com/LAStools/LAStools/releases/tag/v2.0.2). Would it be possible to have another one? Open development has continued, but other things depend on tagged releases, such as inclusion of LAStools in @conda-forge and other packaging efforts. Users in those situations only get updates when releases are provided.

rapidlasso, 2024-04-16:
[Here](https://github.com/LAStools/LAStools/releases/tag/v2.0.3) you are.
In general we recommend to download the current binaries at [rapidlasso download area](https://rapidlasso.de/downloads/).

jmckenna, 2024-04-17:
@rapidlasso also it would be good if all downloads on "rapidlasso download area" and also as part of the GitHub release are tagged with a version in the filename (such as `LAStools-2.0.3.tar.gz` instead of just  `LAStools.tar.gz`), speaking as a packager.