LAStools/LAStools #193: LasreaderLAS::open(std::istream&)

afabri, 2024-04-29:
The `open()` function no longer takes a stream as input and the `LASreadOpener` seems only to take a filename as argument.  What is the recommended way to read from a stream?   We would need that for a CGAL [function](https://github.com/CGAL/cgal/blob/master/Point_set_processing_3/include/CGAL/IO/read_las_points.h#L387) to stay backward compatible.

thbeu, 2024-04-30:
> no longer

What do you mean by "no longer"? Where did you notice the breaking behaviour?

https://github.com/LAStools/LAStools/blob/d815ae254308bf568ad19d3a89862bdca6fa7759/LASlib/inc/lasreader_las.hpp#L69

soesau, 2024-05-02:
Constructing LASreaderLAS with nullptr and then using open() works.
Thank you.

afabri, 2024-05-02:
You are right that `open()`still exists. But the default constructor no longer exists, so we were wondering which `LasReadOpener` to give to the constructor. Apparently `nullptr` does the job.