LAStools/LAStools #196: Extrabytes file corruption? Files created with LAStools cause crashing or cannot be opened in other software packages

beyakk, 2024-05-29:
Fully updated versions of R, rlas, lidr (version 4.1.1). I am working with las files created in Agisoft Metashape with no problem.

If I try to load the same dataset created in LASTools or Whiteboxtools (opencore for both) R crashes consistently, I also cannot open these files with Agisoft Metashape (I get an error message that the format is invalid).

Sounds like this is possibly an issue with LASlib [r-lidar/rlas issue 70](https://github.com/r-lidar/rlas/issues/70)

[test.zip](https://github.com/LAStools/LAStools/files/15477067/test.zip)

I have created very small subset files that show the issue- clip20m.las was created in Agisoft Metashape and lidR and loads just fine. crash-clip20m.las was created in AgisoftMetashape and LASTools lasclip, it cannot be read into lidR with readlas() as it causes an immediate crash of R or RStudio and gets an invalid format error message when trying to import into Agisoft Metashape.

The files are las version 1.2, I am wondering if this might be solved by starting with the original files as las version 1.4 (as it sounds like extrabyte formatting in 1.2 has been a known issue that version 1.4 was designed to solve)? I did try converting one of these to 1.4 using LAStools but with mixed success (I could open it in Agisoft but it still caused R/lidr to crash).

Advice/solutions much appreciated!

manfred-brands, 2024-05-29:
lasinfo for `clip20.las` shows:
```
  point data format:          3
  point data record length:   37
```

accounting for the `3` extra bytes.

lasinfo for `crash-clip20.las` shows:
```
  point data format:          3
  point data record length:   34
```

Despite the VLR code still showing extra bytes, the file actually doesn't seem to contain any.

beyakk, 2024-05-30:
Original comment updated to clarify issue.