LAStools/LAStools #209: blast2dem "EPSG code 1127 not found in 'vertcs.csv' file" error

CamMorin, 2024-09-04:
Hello,

We were using an old version of LASTools and are currently trying to update it to a recent one but we are having difficulties with blast2dem. With data in the vertical datum CGVD2013, we are not able to use blast2dem and get this error and no output is created :

> EPSG code 1127 not found in 'vertcs.csv' file
> set_VerticalCSTypeGeoKey: look-up for 1127 not implemented
> ERROR: unknown vertical EPSG code in '-vertical_epsg 1127'.
> tin2dem usage error. contact 'martin@rapidlasso.com'
> duplicate_points 2099

When looking at the vertcs.csv file, we can find the code 1127, just like in our old version of LASTools.

Here's is an example of a file where we get this error in the newer version and not with our older version : https://download-telecharger.services.geo.ca/pub/elevation/pointclouds_nuagespoints/NRCAN/Lloydminster_2021/AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS.copc.laz 

And the command we pass : 
`D:\path\to\our\lastools\installation\LAStools\bin\blast2dem64.exe -i AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS.copc.laz -step 5 -o AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS_new.tif`

Thank you

jsmoreau, 2024-09-05:
The code 1127 refers to the [Canadian Geodetic Vertical Datum of 2013 (CGG2013)](https://epsg.io/1127-datum). As @CamMorin wrote, it's definition has never been part of the vertcs.csv file. Blast2dem would work with this definition in version 200101 but fails using version 240125.
@rapidlasso Is there a solution for this problem? Is it possible to manually add the 1127 definition?

jsmoreau, 2024-09-05:
@rapidlasso ,
Old versions of gdal would include a csv file (gdal_datum.csv) to contain these definitions. What would be the appropriate file for lastools?
Here is the definition I have:
1127,Canadian Geodetic Vertical Datum of 2013,vertical,"Defined by the equipotential surface W0 = 62,636,856.0 m^2s^-2, which by convention represents the coastal mean sea level for North America.",2013-11-01,,,1061,"Topographic mapping, geodetic survey.","Replaces CGVD28 (datum code 5114) from November 2013.","Geodetic Survey Division, Natural Resources Canada.",OGP,"2013/12/13",2013.064,0,

rapidlasso, 2024-09-05:
This question is answered at the [user group](https://groups.google.com/g/lastools/c/spI0AmKw2dQ). Does this solve your issue?

rapidlasso, 2024-09-05:
For tools not supporting -vertial_epsg .... you can add this line to the .\lastools\bin\serf\geo\vertcs.csv file:
1127,CGVD2013 height,1127,Canadian Geodetic Vertical Datum of 2013,9001,1,0,6499,,

jsmoreau, 2024-09-05:
> For tools not supporting -vertial_epsg .... you can add this line to the .\lastools\bin\serf\geo\vertcs.csv file: 1127,CGVD2013 height,1127,Canadian Geodetic Vertical Datum of 2013,9001,1,0,6499,,

We will need to do some testing. I was not aware that 1127 had been superseded. We might need to add the entry in the vertcs.csv file in order to use older laz files.

CamMorin, 2024-09-05:
> For tools not supporting -vertial_epsg .... you can add this line to the .\lastools\bin\serf\geo\vertcs.csv file: 1127,CGVD2013 height,1127,Canadian Geodetic Vertical Datum of 2013,9001,1,0,6499,,

I tried to add this line in the vertcs.csv file and I still have the same error as before.

jsmoreau, 2024-09-05:
> This question is answered at the [user group](https://groups.google.com/g/lastools/c/spI0AmKw2dQ). Does this solve your issue?

We did some more testing. The usage of las2las with the -set_ogc_wkt option generates a wkt referencing the 1127 datum that blast2dem cannot read afterwards.
Here are the steps I followed:

**Remove the crs definitions from the file**
`D:\path\to\our\lastools\installation\LAStools\bin\las2las64.exe  **-remove_all_vlrs** -i AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS.copc.laz  -o AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS_wo_crs.copc.laz`

**Reset the crs informations**
`D:\path\to\our\lastools\installation\LAStools\bin\las2las64.exe  **-epsg 2956 -vertical_cgvd2013 -set_ogc_wkt** -i AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS_wo_crs.copc.laz  -o AB_Lloydminster2021_20211028_NAD83_CSRS_UTM12_1km_E5090_N59250_CLASS_1127_written_by_las2las.copc.laz`

**Output of the blast2dem command**
`EPSG code 1127 not found in 'vertcs.csv' file
set_VerticalCSTypeGeoKey: look-up for 1127 not implemented
ERROR: unknown vertical EPSG code in '-vertical_epsg 1127'.
tin2dem usage error. contact 'martin@rapidlasso.com'`

jsmoreau, 2024-09-06:
I think part of the issue is because lastools still uses [WKT1 definitions](https://proj.org/en/9.2/development/reference/cpp/cpp_general.html#general_doc_1WKT1). Using the projinfo tool from the PROJ project, the definitions in WKT2 2015 and 2019 do not include the authority code 1127.

**projinfo wkt1**
`projinfo -q -o WKT1:GDAL "EPSG:2956+6647"

COMPD_CS["NAD83(CSRS) / UTM zone 12N + CGVD2013(CGG2013) height",
    PROJCS["NAD83(CSRS) / UTM zone 12N",
        GEOGCS["NAD83(CSRS)",
            DATUM["NAD83_Canadian_Spatial_Reference_System",
                SPHEROID["GRS 1980",6378137,298.257222101,
                    AUTHORITY["EPSG","7019"]],
                AUTHORITY["EPSG","6140"]],
            PRIMEM["Greenwich",0,
                AUTHORITY["EPSG","8901"]],
            UNIT["degree",0.0174532925199433,
                AUTHORITY["EPSG","9122"]],
            AUTHORITY["EPSG","4617"]],
        PROJECTION["Transverse_Mercator"],
        PARAMETER["latitude_of_origin",0],
        PARAMETER["central_meridian",-111],
        PARAMETER["scale_factor",0.9996],
        PARAMETER["false_easting",500000],
        PARAMETER["false_northing",0],
        UNIT["metre",1,
            AUTHORITY["EPSG","9001"]],
        AXIS["Easting",EAST],
        AXIS["Northing",NORTH],
        AUTHORITY["EPSG","2956"]],
    VERT_CS["CGVD2013(CGG2013) height",
        VERT_DATUM["Canadian Geodetic Vertical Datum of 2013 (CGG2013)",2005,
            AUTHORITY["EPSG","1127"]],
        UNIT["metre",1,
            AUTHORITY["EPSG","9001"]],
        AXIS["Gravity-related height",UP],
        AUTHORITY["EPSG","6647"]]]`

**projinfo wkt2 2019**
`projinfo -q -o WKT1:GDAL "EPSG:2956+6647"
COMPD_CS["NAD83(CSRS) / UTM zone 12N + CGVD2013(CGG2013) height",
    PROJCS["NAD83(CSRS) / UTM zone 12N",
        GEOGCS["NAD83(CSRS)",
            DATUM["NAD83_Canadian_Spatial_Reference_System",
                SPHEROID["GRS 1980",6378137,298.257222101,
                    AUTHORITY["EPSG","7019"]],
                AUTHORITY["EPSG","6140"]],
            PRIMEM["Greenwich",0,
                AUTHORITY["EPSG","8901"]],
            UNIT["degree",0.0174532925199433,
                AUTHORITY["EPSG","9122"]],
            AUTHORITY["EPSG","4617"]],
        PROJECTION["Transverse_Mercator"],
        PARAMETER["latitude_of_origin",0],
        PARAMETER["central_meridian",-111],
        PARAMETER["scale_factor",0.9996],
        PARAMETER["false_easting",500000],
        PARAMETER["false_northing",0],
        UNIT["metre",1,
            AUTHORITY["EPSG","9001"]],
        AXIS["Easting",EAST],
        AXIS["Northing",NORTH],
        AUTHORITY["EPSG","2956"]],
    VERT_CS["CGVD2013(CGG2013) height",
        VERT_DATUM["Canadian Geodetic Vertical Datum of 2013 (CGG2013)",2005,
            AUTHORITY["EPSG","1127"]],
        UNIT["metre",1,
            AUTHORITY["EPSG","9001"]],
        AXIS["Gravity-related height",UP],
        AUTHORITY["EPSG","6647"]]]`

Blast2dem seems to handle the WKT2 2019 definition. Maybe this version should be the default for -set_ogc_wkt. The WKT1 could be supported using an extra parameter for legacy purposes.

What do you think @rapidlasso ?

rapidlasso, 2024-09-06:
We will publish a full PROJ integration into LAStools very soon then problems like this will be solved. 
Till then every workaround is fine.

pdean, 2024-09-11:
Licensing issues?

jsmoreau, 2024-09-11:
> Licensing issues?

What do you mean? We do have a valid license.

pdean, 2024-09-11:
> > Licensing issues?
> 
> What do you mean? We do have a valid license.

I was wondering why it's taken rapidlasso so long to integrate proj.  Always would have been the better solution than duplicating the effort and still producing something inferior.

jsmoreau, 2024-09-12:
> > > Licensing issues?
> > 
> > 
> > What do you mean? We do have a valid license.
> 
> I was wondering why it's taken rapidlasso so long to integrate proj. Always would have been the better solution than duplicating the effort and still producing something inferior.

I don't think this is the issue. According to this page, PROJ can be integrated in commercial software:
https://proj.org/en/9.4/about.html#license