LAStools/LAStools #220: LASTools cannot find libtiff.so.6 on Ubuntu20.04

gaborfekete-greehill, 2024-10-08:
In the latest release of LASTools any las* tool complains about the missing `libtiff.so.6` which is now part of `libtiff-dev=^4.5.*` but on `Ubuntu20.04` the latest version for `libtiff-dev` is `4.1.0`.
In the documentation of the installation it says that the version required to run LASTools is:
`libtiff.so.5 => /lib/x86_64-linux-gnu/libtiff.so.5 (0x00007f5701e53000)`
which seems to be incorrect.

How can I use LASTools on Ubuntu20.04 or what can I do to use it, upgrade my Ubuntu version?

sorry, I mistakenly opened an issue on the github repo

rapidlasso, 2024-10-08:
Thanks for the hint. We will provide a new version today to fix this.

rapidlasso, 2024-10-08:
Please use [this build](https://downloads.rapidlasso.de/LAStools.241008.Ubuntu20.tar.gz) for Ubuntu 20.04 LTS (LAStools Version 241008). We will extend our build to support older versions by default.

closed 2024-10-08 by rapidlasso