LAStools/LAStools #223: Missing include files when installing on Linux

EvanZarrella, 2024-11-08:
Currently trying to compile LAStools from source on Linux.
From what I get from the documentation, the process of compiling and installing LAStools should look like :
```
git clone https://github.com/LAStools/LAStools
cd LAStools
cmake .
cmake --build .
make
sudo make install
```

While the library is correctly installed, when trying to compile against it, it finds that `laszip_common.h` is missing meaning that those files are not copied when installing. And I tried installing LASzip then LAStools without much success.

A workaround for anyone interested is to simply copy the files where they are needed : `~/LAStools/LASzip/include/laszip$ sudo cp ./* /usr/local/include/LASlib/`

EvanZarrella, 2024-11-08:
Ok so it's basically what #208 is doing, but now it's got an issue attached

Ylannl, 2024-12-04:
I can confirm this issue.

closed 2024-12-06 by rapidlasso