LAStools/LAStools #224: -proj_epsg losing precision

pdean, 2024-11-08:
I'm trying to convert MLS and static laz files to new coordinate system but
I'm losing precision.  The original coordinates are 0.001 0.001 0.001 but the
output is 0.01 0.01 0.001.

I've created an example to demonstrate by extracting some points from a
testfile

    las2las64 -i in.laz -oparse xyzt -subseq 0 5 -o in.txt 

Here's in.txt

    503619.629 6950590.427 29.373 1371183201.151386
    503629.534 6950595.836 33.205 1371183201.151415
    503619.591 6950590.503 29.402 1371183201.156382
    503629.227 6950595.668 32.843 1371183201.156407
    503619.468 6950590.432 29.100 1371183201.161373

I've converted these using cs2cs

    cs2cs -f "%.3f" epsg:7856 epsg:3113 in.txt >conv.txt

Here's conv.txt

    53620.446	147810.053 29.373 1371183201.151386
    53630.355	147815.465 33.205 1371183201.151415
    53620.408	147810.129 29.402 1371183201.156382
    53630.048	147815.296 32.843 1371183201.156407
    53620.285	147810.058 29.100 1371183201.161373

Then I've used las2las to make a las file, convert it, then dump the coords as
text

    las2las64 -i in.txt -iparse xyzt -rescale 0.001 0.001 0.001 -epsg 7856 -o in.las
    las2las64 -i in.las -proj_epsg 7856 3113 -o out.las 
    las2las64 -i out.las -oparse xyzt -o out.txt 

Here's out.txt

    53620.45 147810.05 29.373 1371183201.151386
    53630.36 147815.46 33.205 1371183201.151415
    53620.41 147810.13 29.402 1371183201.156382
    53630.05 147815.30 32.843 1371183201.156407
    53620.29 147810.06 29.100 1371183201.161373

All the horizontal coordinates have been rounded to nearest centimetre.  Our scanners are better than that!

TobBei, 2024-11-13:
Hello,

Thank you for pointing this out. The '-rescale' argument previously had no effect when a CRS transformation was applied. 
This has now been corrected, and the adjustment will be included in the next release.

Best regards

rapidlasso, 2024-12-06:
[fixed](https://github.com/LAStools/LAStools/commit/6b246d1af5a13bf9feebf9697b6d928836dc3c07)