LAStools/LAStools #225: -proj_epsg not working with wildcards

pdean, 2024-11-08:
I'm converting a directory of laz file to different cs.  But only the first one
converts, the rest pass through unchanged.

I've created a test directory with 3 small files all the same, in1.las in2.las
in3.las.  These were created from a text file

    503619.629 6950590.427 29.373 1371183201.151386
    503629.534 6950595.836 33.205 1371183201.151415
    503619.591 6950590.503 29.402 1371183201.156382
    503629.227 6950595.668 32.843 1371183201.156407
    503619.468 6950590.432 29.100 1371183201.161373 

I then run this

    las2las64 -i *.las -proj_epsg 7856 3113 -olas -odir bcsg -odix _bcsg
    las2las64 -i bcsg/*.las -oparse xyzt -otxt -odir bcsg

Apart from losing precision, bcsg/in1_bcsg.txt is correct

    53620.45 147810.05 29.373 1371183201.151386
    53630.36 147815.46 33.205 1371183201.151415
    53620.41 147810.13 29.402 1371183201.156382
    53630.05 147815.30 32.843 1371183201.156407
    53620.29 147810.06 29.100 1371183201.161373

But in2 and in3 convert to this, which is the same as the input

    503619.629 6950590.427 29.373 1371183201.151386
    503629.534 6950595.836 33.205 1371183201.151415
    503619.591 6950590.503 29.402 1371183201.156382
    503629.227 6950595.668 32.843 1371183201.156407
    503619.468 6950590.432 29.100 1371183201.161373

I've had the same problem with lasdatum.  I can get around it by creating a loop
in a batch file.  But I shouldn't have to.  The conversion to text works fine
with wildcards.

TobBei, 2024-11-13:
Hello,

This is indeed a bug, which we have now fixed. 
With the next release, the transformation should be applied to all input files.

Best regards

rapidlasso, 2024-12-06:
[fixed](https://github.com/LAStools/LAStools/commit/6b246d1af5a13bf9feebf9697b6d928836dc3c07)