LAStools/LAStools #226: -proj_string unrecogised format

pdean, 2024-11-10:
I'm needing to convert a directory of laz files to another coordinate system using a 2d helmert transformation.  I've determined the parameters using the least squares formula on coordinate files we have in both systems.  I can check the proj string is correct using cct and it works perfectly viz.

    cct +proj=helmert +convention=coordinate_frame +x=-207.07998 +y=-6002831.790225 +s=1.00040759214702 +theta=-0.0366579097505624 gda.txt

But when I try the same string with las2las I get this output

    C:\Users\peter\Desktop\test>las2las64     -i in.laz     -proj_string "+proj=helmert +convention=coordinate_frame +x=-207.07998 +y=-6002831.790225 +s=1.00040759214702  +theta=-0.0366579097505624"     -o out.laz
    proj_create: unrecognized format / unknown name
    ERROR: Failed to create the CRS objects using PROJ string: Invalid PROJ string syntax
    
What am I missing here?

TobBei, 2024-11-13:
Hello,

The error message does not provide the correct feedback that it should, thanks for the comment. 
The proj string appears to be fine. 
However, currently, only transformations between different Coordinate Reference Systems (CRS) are supported. 
This means that the CRS information must be included in the proj string. 
Pure Coordinate System (CS) operations such as ‘+proj=helmert’ are not supported at the moment.

Best regards

pdean, 2024-11-13:
Can you please support them.  At the moment there seems to be no way to do a 2d helmert.  See issue #227.  

-proj_string [s] [t]                : (For experienced users) uses the PROJ lib to perform a CRS transformation. Optionally, the source CRS [s] can be specified using PRO string (deafult from the input file header). In addition, the target CRS [t] must be specified using PROJ string.  **Furthermore a single PROJ string [s] can also be specified, which directly describes a transformation or operation**

pdean, 2024-11-13:
And further, terrascan doesn't include any CRS info in header,  therefore it shouldn't matter there's none in the output.

    lasinfo (241017) report for 'in.laz'
    reporting all LAS header entries:
      file signature:             'LASF'
      file source ID:             0
      global_encoding:            1
      project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
      version major.minor:        1.4
      system identifier:          'ML;'
      generating software:        'TerraScan'
      file creation day/year:     291/2024
      header size:                375
      offset to point data:       375
      number var. length records: 0
      point data format:          7
      point data record length:   36
      number of point records:    0
      number of points by return: 0 0 0 0 0
      scale factor x y z:         0.0001 0.0001 0.0001
      offset x y z:               500000 6900000 -0
      min x y z:                  503531.4300 6950564.4640 14.0810
      max x y z:                  503631.9220 6950612.4800 35.8140
      start of waveform data packet record: 0
      start of first extended variable length record: 0
      number of extended_variable length records: 0
      extended number of point records: 6016281
      extended number of points by return: 5894305 117458 4402 116 0 0 0 0 0 0 0 0 0 0 0
    LASzip compression (version 3.4r3 c3 50000): POINT14 3 RGB14 3
    reporting minimum and maximum for all LAS point record entries ...
      X            35314300   36319220
      Y           505644640  506124800
      Z              140810     358140
      intensity           2      65535
      return_number       1          4
      number_of_returns   1          5
      edge_of_flight_line 0          0
      scan_direction_flag 0          0
      classification      1          4
      scan_angle_rank  -128        127
      user_data           2          2
      point_source_ID   218        514
      gps_time 1371183201.146390 1372905768.562652
      Color R 0 65280
            G 0 65280
            B 0 65280
      extended_return_number          1      4
      extended_number_of_returns      1      5
      extended_classification         1      4
      extended_scan_angle        -21333  21167
      extended_scanner_channel        2      2
    WARNING: there is coordinate resolution fluff (x10) in XYZ
    number of first returns:        5894305
    number of intermediate returns: 4848
    number of last returns:         5889099
    number of single returns:       5771971
    overview over extended number of returns of given pulse: 5771971 230027 13766 514 3 0 0 0 0 0 0 0 0 0 0
    histogram of classification of points:
             1566715  unclassified (1)
             2971097  ground (2)
              502873  low vegetation (3)
              975596  medium vegetation (4)